Klipper3d/klipper #928: [Feature Request] Change feedrate from main screen

jannoke, 2018-11-24:
Could we have rotary encoder on main screen to change the feedrate just as in marlin. Just by scrolling it. I think marlin has some "dead scrolling" in the beginning so that just few ticks won't start to change the speed straight away. 

It's currently not assigned to anything and if there is trouble then it's good to change it, fast. Also since i'm always testing out the limits of firmware and printer then it's something I use often.

drzejkopf, 2018-11-27:
Tune>Speed, really easy to find :D

jannoke, 2018-11-28:
* click...try to scroll as fast as possible to tune
* click... scroll again and try to aim "speed"
* click again..set speed.. 
and not sure if it needs to be clicked again or will it work on the fly.

As opposed to just "scrolll". 

Also  - tune menu is not available out of print. So it can't be preset.

Like I said - it's not like it's robbing away or replacing some other features that are already present and also it's present in Marlin so it's not like I woke up in the morning and thought that it's a good idea all by myself.

KevinOConnor, 2018-12-09:
This type of functionality would require a developer to implement (and/or to create the necessary menu config files).  I'm inclined to close this unless someone is actively working on implementing it.

-Kevin

closed 2019-01-10 by KevinOConnor