Klipper3d/klipper #930: Can't start service

varazir, 2018-11-25:
When I run the command pi@octopi:~/klipper $ sudo service klipper start  I get 
Failed to start klipper.service: Unit klipper.service not found.

No log file 

Linux octopi 4.19.2-v7+ #1162 SMP Thu Nov 15 12:28:05 GMT 2018 armv7l
OctoPrint version : 1.3.10rc2
OctoPi version    : 0.15.0

varazir, 2018-11-26:
Planing to reflash OctoPi

gdachs, 2018-11-26:
As Klipper is not part of OctoPi it makes no sense to reflash with OctoPi. It will not solve the problem. It looks like you just made something wrong during install of Klipper. Repeat that.

Generally, this is Linux, not Windows. Linux nearly never need a reinstall to fix problems.

varazir, 2018-11-26:
> As Klipper is not part of OctoPi it makes no sense to reflash with OctoPi. It will not solve the problem. It looks like you just made something wrong during install of Klipper. Repeat that.
> 
> Generally, this is Linux, not Windows. Linux nearly never need a reinstall to fix problems.

I know linux is much better to handle changes,  I ran the installation script 3 times. 
I upgraded the kernel and the whole OS and now /boot is full. 

Going to do a reflash as I messed up the system.

varazir, 2018-11-26:
Notice now that it took much longer to run the script. I guess when apt failed with setting up the new kernel it exited the script.

KevinOConnor, 2018-11-30:
It sounds like this issue was resolved, so I'm going to close it.

-Kevin

closed 2018-11-30 by KevinOConnor