Klipper3d/klipper #932: Bltouch probe problems

nandinho84, 2018-11-26:
I've been some problems to put my bltouch to work (as a probe, not endstop).
Now, my problem is that the XY axis goes to positions that is not supposed to go.
I have this setup:
[quad_gantry_level]
gantry_corners:
	-90,-30
	375,330
points:
	60,60
	60,240
	240,240
	240,60

The first two probing points seems accurate, then the third probing point it not probing on the 240,240 as it should...  it's probing outside the bed (something more like 200,300). As it probes outside the bed, klipper enters into an error state and the G32 fails.
Any insights of the problem might be?
Here is my log file (p.s. I can't seem to find my klippy.log file, so I've attached the octoprint log file... total noob on klipper...)

[octoprint.zip](https://github.com/KevinOConnor/klipper/files/2613910/octoprint.zip)

BlackStump, 2018-11-26:
have a read as it might help and it is important for any support
https://github.com/KevinOConnor/klipper/blob/master/docs/Contact.md

if you have Windows OS here is a simple scp utility
https://winscp.net/eng/download.php

cosminr86, 2018-11-26:
You must use nozzle position when probing, but must keep in mind tour offset.
If tour bed is 240 and your probe have +20 offset, you must probe at 220.

nandinho84, 2018-11-27:
Turns out, that it was a loosen pulley that was giving this problems... Sorry for the rookie mistake.

closed 2018-11-27 by nandinho84