Klipper3d/klipper #934: How to Calibrate Steps ?

xsasx, 2018-11-26:
How do i calibrate my steps with klipper - in marlin it was easy and many tutorials are around - but how it goes with klipper `?

theopensourcerer, 2018-11-26:
Do the same calibration as with ANY printer and then divide the result by 1 to get mm/step rather steps/mm.

xsasx, 2018-11-26:
ok - and where i put my result in ?

theopensourcerer, 2018-11-26:
Err, look at the example configuration files. They will show you all the options and are well commented. You need to set the step_distance parameter, e.g.

https://github.com/KevinOConnor/klipper/blob/master/config/example.cfg#L33

xsasx, 2018-11-26:
mhhhh i mean extrusion steps - so i messure my filament- extrude 10*10 got 21mm left and than ? sorry im not sooooo familiar with all these things

theopensourcerer, 2018-11-26:
There are lots of "howtos" online about how to do this:

https://mattshub.com/2017/04/19/extruder-calibration/

I am not sure what else you really need to know... The extruder configuration is also in the same example file:

https://github.com/KevinOConnor/klipper/blob/master/config/example.cfg#L94

xsasx, 2018-11-26:
If  i want to sen M503 it says command does not exist - thats my problem - if i follow one of these how tos it wont work with klipper that way :/ so i want to know how i can calibrate my esteps under klipper

theopensourcerer, 2018-11-26:
Why do you need to send M503? There is no "EEPROM" in klipper. Current settings are in your printer.cfg file.

ikarisan, 2018-11-26:
M503 is not needed to calibrate the steps. You will find your current values (step_distance:) inside the printer.cfg file.

gdkinney, 2018-11-26:
Most online tutorials are assuming step/mm (used by Marlin and others) instead of mm/step used by Klipper. You need to invert the measured/expected ratio, in this case it's (90/100)*0.0105=0.00945 .

ikarisan, 2018-11-26:
@xsasx I would calc it like this:

1) Mark you filament 120mm above the entry to your extruder.
2) Heat up the nozzle to your desired printing temperature
3) Home all axis to get in "printer ready" state
4) Lift up your nozzle by 50mm (to make room for the filament!)
5) Execute the following command (one by one)
  5a) G92 E0
  5b) G1 E100 F100
6) Measure the distance between your extruder entry and the mark on your filament.

I.e if it is 28mm instead of 20mm (120mm - 100mm) than you are UNDERextruding by 8mm ==> 92mm instead of 100mm.

Now calc:

c := current value in configuration.cfg
m := measured left over filament
d := desired mm
n := new value for configuration.cfg

((120 - m) / d) * c = n
((120 - 28) / 100) * 0.010500 = 0,009660
(92 / 100) * 0.010500 = 0,009660

As you can see, for underextrusion the new value is lower than the old one. 
You may play around with the last two numbers to fine tune.

KevinOConnor, 2018-12-09:
It looks like the original question was answered, so I'm going to close this issue.

-Kevin

closed 2018-12-09 by KevinOConnor

Ianmcmill, 2018-12-18:
I still don't get it why Klipper switches the whole quasi standard of steps/mm into mm/steps. But then again I am free to use Klipper or let it be. Am I right ;)

KevinOConnor, 2018-12-18:
FYI, it's done that way to use consistent measurement units in the config file - mm for distance, amps for current, volts for voltage, seconds for time, ohms for resistance, etc.  I know it's different from other firmware, but it's just weird to configure this one parameter in units of inverse millimeters.

-Kevin

dotps1, 2018-12-18:
i agree with the consistency, I put a note in my config file so I don't forget:

```
# Measure 120mm, then extrude 100mm
# G1 E100 F100
# step_distance = old_e_steps * ((120 - distance_to_mark) / 100)
```

NameOfTheDragon, 2020-10-22:
> @xsasx I would calc it like this:
> 
> 1. Mark you filament 120mm above the entry to your extruder.
> 2. Heat up the nozzle to your desired printing temperature
> 3. Home all axis to get in "printer ready" state
> 4. Lift up your nozzle by 50mm (to make room for the filament!)
> 5. Execute the following commands (one by one)
>    5a) G92 E0
>    This resets the "extruded material" value to 0.
>    5b) G1 E100 F100
>    This extrudes 100mm filament with 100mm/min.
> 6. Now measure the distance between your extruder entry and the mark on your filament.
> 
> I.e if it is 28mm instead of 20mm (120mm - 100mm) than you are UNDERextruding by 8mm ==> 92mm instead of 100mm. If it shows 15mm than your are OVERextruding by 5mm ==> 105mm.
> 
> Now calc:
> 
> c := current value in configuration.cfg
> m := measured left over filament
> d := desired mm
> n := new value for configuration.cfg
> 
> ((120 - m) / d) * c = n
> ((120 - 28) / 100) * 0.010500 = 0,009660
> (92 / 100) * 0.010500 = 0,009660
> 
> As you can see, for underextrusion the new value is lower than the old one.
> You may play around with the last two numbers to fine tune.

It would be absolutely awesome for people new to Klipper if this appeared in the "getting started" document. It is quite a culture shock coming from Marlin and this would be a great help, in my opinion.

Hairyloon, 2021-05-04:
My Pi broke, so I've had to reinstall and I see that we've moved from steps/mm, which was simple and easy to fathom, to microsteps and rotation distance, which appears to be based on guesswork...

> If unsure, set microsteps: 16 in the config and use 16 in the formula above.

I can imagine that maybe this makes it easier to programme since the calculations have to be done by the user rather than the software, and maybe there are some additional instructions that I've missed: perhaps a command to move (n) microsteps, so we can see how far it goes, but otherwise it seems to make no sense...

manyhats-mike, 2021-05-05:
Yeah, it wasn't obvious about the transition from step_distance to rotation_distance and this is the #1 google hit.  A bit of digging, I'm hoping I can follow this guide for anyone else that lands here first:
https://github.com/KevinOConnor/klipper/blob/master/docs/Rotation_Distance.md