Klipper3d/klipper #935: EZABL issue

SynMaki, 2018-11-26:
Hey All,

Im currently trying to install a EZABL on my CR-10S.

It took me several hours,tried different configs, just to notice that is seems like the Z is not triggering . always hit the bed...

What ive just learned:If  ill hold my hand to the inductior, start a M119 command and the Z always seems not to be triggered even with the smallest config i could test

When i do the same with the printersave.txt config, the Z can be triggered correctly, so i dont think i got a broken ABL sensor

What did i wrong here?

[klippy.log](https://github.com/KevinOConnor/klipper/files/2617444/klippy.log)
[printer.txt](https://github.com/KevinOConnor/klipper/files/2617445/printer.txt)
[printersave.txt](https://github.com/KevinOConnor/klipper/files/2617446/printersave.txt)

SynMaki, 2018-11-28:
Had some help and noticed, i need to change the wires on the Z Cable. Case Closed. Not a klipper "issue"

closed 2018-11-28 by SynMaki

timryder, 2019-01-01:
Do you have a config file for the EZABL Sensor to work on Klipper?  I also have one and am stumped for now.

dbrinas, 2021-06-11:
> Do you have a config file for the EZABL Sensor to work on Klipper? I also have one and am stumped for now.

did you ever get a solution for the ezabl sensor to work on klipper??

dbrinas, 2021-06-11:
> Had some help and noticed, i need to change the wires on the Z Cable. Case Closed. Not a klipper "issue"

what was your solution to getting the ezabl to work with klipper. pls post a config file for the probe  and z. thanks.