Klipper3d/klipper #936: Print job cancel on every little error

CageFox, 2018-11-26:
[image]

On every little error klipper cancels whole print, is there settings to make this errors just warnings?

Exceeded max acceleration - set max available, show warning and continue print. Exceeds max/min X or Y or Z? Just move to maximum/minimum possible show warning and continue the print. Thats what i want, way to compensate my mistakes, not to punish me

nandinho84, 2018-11-27:
I would suggest that klipper would show a warning message before the print job starts, with the option to cancel the job or to set the values at the max/min supported by the firmware.

CageFox, 2018-11-27:
It happened in middle of print. My mistake, but maybe it will be better not cancel entire print

nandinho84, 2018-11-27:
I understand what you mean... It would be great if klipper could make an analysis of the g-code prior to start the print and give some warnings...

CageFox, 2018-11-27:
The command was entered in terminal by hands ;) 
I just forgot i have 1200 maximum acceleration for this print and tryed to set 1600. Little harmless mistake costs me whole print

I think klipper a little too strict in whole, there is very small number of situations which requires print terminate. Most errors are "Warn and continue"

I think there can be section in config:
[error_handling]
bounds_errors=[ignore|warn|terminate]

BlackStump, 2018-11-28:
@CageFox  have you tried to change Octoprints actions on a error? not sure if it would satisfy your requirements and would be at your risk.
[image]

jakep82, 2018-11-28:
Yeah, I believe this is actually Octoprint doing this.  I dont' believe Klipper actually cancels prints as your error indicates.  If something significant happens it will shutdown the MCU which requires a firmware_restart.  In your screenshot it appears that you simply have Octoprint set to cancel prints whenever a firmware error is reported.  I have mine set to ignore errors because Klipper with shut itself down if it's a critical error.

CageFox, 2018-11-28:
Thanx, guys, it was not very obviuos place to search. I check this and close issue

CageFox, 2018-11-28:
By the way, suppose klipper receives G0 command to move to X400 when current position 100 and max position 300. What will klipper do besides the issuing an Error? Move to 300? Or stay at 100?

fritzw, 2018-11-29:
@CageFox: According to my experience it will silently ignore move commands with illegal coordinates and stay where it was. Why don't you just try it yourself using the OctoPrint console?

CageFox, 2018-11-29:
I was in the middle of print and can not do this

Now I checked and found strange things

Max y for my printer is 320. If current Y is 200 and G0 Y400 printer throws and error and moves to 300. If the current y is 290 and G0 Y400 no move

Strange movements: not fully ignored and not move to bound

BlackStump, 2018-11-29:
I am struggling to understand the situation that anyone would want to send a command outside of printer constraints.
If you set your dimensions of your printer bed into your slicer it will not slice a model larger then that!
if you slice a model that is smaller then your bed then how do you get a movement command that is larger then your bed?
if you are manually issuing commands that are larger then your Firmware or Octoprints constrains then the firmware should issue a error this is right and correct.
If you have your bed dimensions entered into Octoprint I think Octoprint will also issue a error and there needs to be a understanding that Octoprint is not Klipper. Octoprint has its own set of error behaviour settings.

CageFox, 2018-11-29:
> if you slice a model that is smaller then your bed then how do you get a movement command that is larger then your bed?

Bounds is not always bounds of print volume. Acceleration for example bounded too. Feed rate. Flow rate.

Pause g code script can be comfortably maked using g91 relative mode. In relative mode G0 X-10 can throw all by moving out of lower bound

For example my typical pause print script contained G0 Y+200. Marlin silintly moved to bound or to Y+200 but klipper throws and moves strange

I lost one print when specified acceleration larger then bound

BlackStump, 2018-11-29:
Marlin is a mature firmware been around for a number of years but who wants to use it when Klipper is better and that is why you are here yes.
Klipper is not Marlin or Repetier and I for one would not want Klipper doing what Marlin does just because it is Marlin.
The Acceleration you could put a higher value in your config and add then set your particular accel in your slicer start code if you have a different accel for different models.
Again not sure why you would want to do that because if you fine tune your printer for quality then usually there is a limit for accel any higher and print quality suffers or you enter into resonance issues.
For me I find that limit and that is the limit I enter into the config and never change unless there has been a hardware change then I fine tune again.
If you are referring to you original post you lost a print because of Octoprints bounds.

CageFox, 2018-11-29:
> Klipper is not Marlin or Repetier and I for one would not want Klipper doing what Marlin does just because it is Marlin

I think it will be good if klipper can do all Marlin can do, but better

> Again not sure why you would want to do that because if you fine tune your printer for quality

The process of fine tuning not completed yet, and sometimes i want to change something in runtime and look what will change

You speak "Marlin is a mature firmware been around for a number of years" and that's true, but this also means Marlin's behaviors are mature and became this mature thru tries, attempts, blood and fails. 

I think this became a little offtopic and this issue can be closed. Thanx for point me in right direction!

CageFox, 2018-12-07:
I need to reopen this issue. Today print stoped by klipper, not by octoprint:

>  0:30:11.627: Firmware was halted, trying to reconnect. Eventually running print is stopped.
> 0:30:12.680: !! Error on 'SET_VELOCITY_LIMIT MAX_ACCEL=1600 SQUARE_CORNER_VELOCITY=17': SQUARE_CORNER_VELOCITY must have maximum of 10.0

As always it's my mistake I forget i set square velocity to 10 in config and started script with value 17

But i see no reason to halt and to cancel print! Can it be changed?

AxMod3DPrint, 2018-12-08:
I'm guessing the answer will be no, because it is a safety feature, with
settings like that it is likely Klipper will eventually crash as it would
not be able to keep up from my understanding, although I'll let others chip
in here.

BlackStump, 2018-12-08:
A config is where you put your limits, imho why would you want klipper to ignore those limits when you enter them. That seems counter intuitive.

CageFox, 2018-12-08:
I think the current print is prime priority. If command can be safely ignored it must not break the print. Command that set corner velocity CAN be safely ignored

Counter intuitive for me broken print because of this

trevjonez, 2018-12-08:
I agree attempting to set past limit should probably log a warning and apply the limit value.

CageFox, 2018-12-08:
> attempting to set past limit should probably log a warning and apply the limit value

Yes, only this approach is "intuitive"

BlackStump, 2018-12-08:
While what you suggest is idea  ie  have the firmware cover every aspect of user error. There is the option for you to submit a pr with that work to do as you suggested . Just saying easy to forget that Kevin is providing this firmware off of his hard work !
 When you say the print is of priority then maybe a understanding of self imposed limits and not sending a command that is above those limits should also be a priority?
 I very much doubt that increasing square corner velocity from 10 to 17 was a print improving request.
Ultimately Kevin will decide what is important not me just expressing a opinion on this matter.

CageFox, 2018-12-08:
> I very much doubt that increasing square corner velocity from 10 to 17 was a print improving request

It also very counter intuitive such harmless attempt can ruin the whole print

> understanding of self imposed limits and not sending a command that is above those limits should also be a priority?

I think if firmware can make user errors less destructive it MUST do this. This is the computer's work, to help us, not to harm us and not to punish us.

KevinOConnor, 2018-12-09:
This isn't something I plan to work on.  That said, if someone wishes to submit a code change so that SET_VELOCITY_LIMIT places a ceiling on values instead of reporting an error then that is fine.  (For example, as the M204 command does.)

-Kevin

KevinOConnor, 2019-01-04:
FYI, the SET_VELOCITY_LIMIT command should no longer report an error on a parameter that is too high (it will use the maximum valid value in that case) - commit a2f0c36e.

-Kevin

closed 2019-01-04 by CageFox