Klipper3d/klipper #938: [homing_override] does break homing altogether

uuzi, 2018-11-27:
I can't seem to get homing override to work. Since my z probe is offset from the nozzle, I need z homing to go to the middle of the bed before homing.
I've tried several things like just enabling homing override for z (axes: z).
Everything I tried results in a timeout while homing x.

When I comment out the homing override, x and y home just fine.

Any help is much appreciated.
[klippy.log](https://github.com/KevinOConnor/klipper/files/2618392/klippy.log)

rgrae81, 2018-11-27:
HI, 
I use this one:

`[homing_override]
set_position_z: 5
axes: z
gcode:
 G90   ; Uncomment these 2 lines to blindly lift the Z 2mm at start
 G1 Z7 F600   ; Move Z up a bit to prevent scratching
 G28 X0 Y0
 G1 X150 Y150 F3600
 G28 Z0`

uuzi, 2018-11-28:
Thank you for your quick reply!

I will try these settings once I come home tonight.

technoo10201, 2020-02-16:
> HI,
> I use this one:
> there must be one space before each g-command (this editor here is deleting them)
> 
> [homing_override]
> set_position_z: 5
> axes: z
> gcode:
> G90 ; Uncomment these 2 lines to blindly lift the Z 2mm at start
> G1 Z7 F600 ; Move Z up a bit to prevent scratching
> G28 X0 Y0
> G1 X110 Y50 F3600
> G28 Z0

Thanks ! It helped me and now it works !