Klipper3d/klipper #941: Issues installing on Ender 3

Gulrokacus, 2018-11-28:
Hi, so I've tried a few different ways to install Klipper on an ender 3. I attempted using the ender 3 config file command as well. No matter what I'm receiving errors with the programmer responding. 

```
------------------------------------------------------------------------------
OctoPrint version : 1.3.9
OctoPi version    : 0.15.1
------------------------------------------------------------------------------

pi@octopi:~ $ cd ~/klipper/
pi@octopi:~/klipper $ make menuconfig
scripts/kconfig/mconf /home/pi/klipper/src/Kconfig

*** End of the configuration.
*** Execute 'make' to start the build or try 'make help'.

pi@octopi:~/klipper $ make
  Build Kconfig config file
  Creating symbolic link out/board
  Compiling out/src/sched.o
  Compiling out/src/command.o
  Compiling out/src/basecmd.o
  Compiling out/src/debugcmds.o
  Compiling out/src/gpiocmds.o
  Compiling out/src/stepper.o
  Compiling out/src/endstop.o
  Compiling out/src/adccmds.o
  Compiling out/src/spicmds.o
  Compiling out/src/thermocouple.o
  Compiling out/src/i2ccmds.o
  Compiling out/src/pwmcmds.o
  Compiling out/src/lcd_st7920.o
  Compiling out/src/lcd_hd44780.o
  Compiling out/src/buttons.o
  Compiling out/src/tmcuart.o
  Compiling out/src/avr/main.o
  Compiling out/src/avr/timer.o
  Compiling out/src/avr/gpio.o
  Compiling out/src/avr/adc.o
  Compiling out/src/avr/spi.o
  Compiling out/src/avr/i2c.o
  Compiling out/src/avr/hard_pwm.o
  Compiling out/src/avr/watchdog.o
  Compiling out/src/avr/serial.o
  Compiling out/src/generic/serial_irq.o
  Building out/compile_time_request.o
Version: v0.6.0-645-gb8f92c7-20181128_033842-octopi
  Linking out/klipper.elf
  Creating hex file out/klipper.elf.hex
pi@octopi:~/klipper $ ls -l /dev/serial/by-id/
total 0
lrwxrwxrwx 1 root root 13 Nov 28 03:31 usb-FTDI_FT232R_USB_UART_A506VHCF-if00-port0 -> ../../ttyUSB1
pi@octopi:~/klipper $ make flash FLASH_DEVICE=/dev/serial/by-id/usb-FTDI_FT232R_USB_UART_A506VHCF-if00-port0
  Flashing out/klipper.elf.hex to /dev/serial/by-id/usb-FTDI_FT232R_USB_UART_A506VHCF-if00-port0 via avrdude
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x00
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 2 of 10: not in sync: resp=0x00
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 3 of 10: not in sync: resp=0x00
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 4 of 10: not in sync: resp=0x00
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 5 of 10: not in sync: resp=0x00
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 6 of 10: not in sync: resp=0x00
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 7 of 10: not in sync: resp=0x00
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 8 of 10: not in sync: resp=0x00
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 9 of 10: not in sync: resp=0x00
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 10 of 10: not in sync: resp=0x00

avrdude done.  Thank you.

src/avr/Makefile:32: recipe for target 'flash' failed
make: *** [flash] Error 1
```

manu7irl, 2018-11-28:
> I solved my own problem. I didn't have a bootloader installed despite
> upgrading the firmware to Marlin.
>
> Seems to work now, it installed. Thank you for letting me post!

good luck to you.
can you come to the discord channel to share your first print results?
I also got an ender3 with all sorts of printing problems.
https://discord.gg/ZuGpss
thanks.

KevinOConnor, 2018-11-30:
It sounds like this issue was resolved, so I'm going to close it.

closed 2018-11-30 by KevinOConnor