Klipper3d/klipper #942: Unable to Translate Pin - config_soft_pwm ....

Gulrokacus, 2018-11-28:
Your printer's firmware reported an error. Due to that OctoPrint will disconnect. Reported error: Unable to translate pin name: config_soft_pwm_out oid=9 pin=ar32 cycle_ticks=320000 value=0 default_value=0 max_duration=0

[klippy.log](https://github.com/KevinOConnor/klipper/files/2622991/klippy.log)

I'm using the default ender 3 profile..

BlackStump, 2018-11-28:
The Ender 3  has a Melzi board and by default doesn't have a connection for a bltouch unless you do some hacking into the lcd ribbon to get access to unused pins. 
If you have done that are you sure you have the right pins named in your config?
The default Ender3 printer.cfg does not use the bltouch.
If you do a search there is info on how to attach a probe to the Ender3 and maybe the pins used to do that.

Gulrokacus, 2018-11-28:
Yep, I have the right pins and hacked into the LCD ribbon. I'm currently trying to get the Klipper software to run without BLTouch and then I'll add it in. So currently its not running with the...

"Your printer's firmware reported an error. Due to that OctoPrint will disconnect. Reported error: Unable to translate pin name: config_soft_pwm_out oid=9 pin=ar32 cycle_ticks=320000 value=0 default_value=0 max_duration=0" error.

Gulrokacus, 2018-11-29:
So I changed the pins to 27, still can't seem to get it right. I might change to a MKS l instead of the board I have. Seems other ender 3'rs are having better luck with that board and BLtouch.

KevinOConnor, 2018-11-29:
If you wish to use the alias "ar32" (or similar) then you need to add "pin_map: arduino" to the [mcu] section of your config.  See: https://github.com/KevinOConnor/klipper/blob/master/docs/FAQ.md#how-do-i-convert-a-marlin-pin-number-to-a-klipper-pin-name

closed 2018-12-09 by KevinOConnor