Klipper3d/klipper #943: After canceling in Octoprint all axis have to be homed

ikarisan, 2018-11-28:
Hi!

Sometime I have to cancel a print by pressing the button in Octoprint. After finishing the last movement the printer stops. Thats okay. But now I have to home all axis again before I can move the nozzle off the way. But I cannot home with an object on the heat bed and I cannot remove the object while the hot nozzle is stuck on it. :( 

These are the corresponding entries of the log:

`2018-11-27 18:22:16,933 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2018-11-27 18:30:57,981 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2018-11-27 18:30:59,244 - octoprint.plugins.tracking - INFO - Sent tracking event ping, payload: {}
2018-11-27 18:36:15,021 - octoprint.util.comm - INFO - Changing monitoring state from "Printing" to "Cancelling"
2018-11-27 18:36:18,771 - octoprint.util.comm - INFO - Changing monitoring state from "Cancelling" to "Operational"
2018-11-27 18:36:22,474 - octoprint.util.comm - WARNING - Received an error from the printer's firmware: Must home axis first: 161.363 101.784 21.400 [824.580]
| Last lines in terminal:
| Send: N2443 M108*27
| Recv: ok
| Send: N2444 M84*41
| Recv: // Unknown command:"M108"
| Recv: ok
| Send: N2445 M104 T0 S0*22
| Recv: ok
| Send: N2446 M140 S0*81
| Recv: ok
| Send: N2447 M105*18
| Recv: ok
| Send: N2448 M106 S0*93
| Recv: ok T0:235.0 /0.0 B:74.9 /0.0
| Changing monitoring state from "Cancelling" to "Operational"
| Recv: ok
| Send: G91
| Send: G1 Z10 F200
| Recv: ok
| Recv: !! Must home axis first: 161.363 101.784 21.400 [824.580]
| Send: G90`

jakep82, 2018-11-28:
You need to change the settings in Octoprint.  By default it will issue an M84 after canceling a print which turns the motors off.  If you delete that line from the cancel print settings, the motors will not be turned off, and you can still move the head.

ikarisan, 2018-11-28:
Ah! Thank you very much for this hint.

arbitur, 2018-12-06:
> You need to change the settings in Octoprint. By default it will issue an M84 after canceling a print which turns the motors off. If you delete that line from the cancel print settings, the motors will not be turned off, and you can still move the head.

Could you also tell us specifically which setting this is?

jakep82, 2018-12-06:
Printer>GCODE Scripts>After print job is cancelled

KevinOConnor, 2018-12-09:
I'm going to close this as it looks like the original question was answered.  Also, note the FAQ entry for this: https://github.com/KevinOConnor/klipper/blob/master/docs/FAQ.md#why-cant-i-move-the-stepper-before-homing-the-printer

-Kevin

closed 2018-12-09 by KevinOConnor