Klipper3d/klipper #945: ADC five buttons keyboard control

ZhenyaKa, 2018-11-30:
How to configure 5 buttons keyboard?
Or it not realized?
[http://ipic.su/img/img7/fs/kiss_386kb.1543542783.png](url)

KevinOConnor, 2018-11-30:
I suspect this is a similar request to #898 (and #609).  It requires development work in order to support buttons on an analog sensor.

-Kevin

KevinOConnor, 2018-11-30:
I do not have any immediate plans to work on this.  Others have expressed interest in developing support, but I do not know if they are working on it.

-Kevin

ZhenyaKa, 2018-11-30:
> I do not have any immediate plans to work on this. Others have expressed interest in developing support, but I do not know if they are working on it.
> 
> -Kevin

Can you add range reading for ADC input? ar12(min, max)
And it will be enouth.

KevinOConnor, 2018-12-04:
Lets close this issue in favour of the existing issue #898 / # 609.

closed 2018-12-04 by KevinOConnor