Klipper3d/klipper #946: Virtual SD card print fails when initiated from the printer

akpch, 2018-11-30:
[log-failed-sdcard.log.gz](https://github.com/KevinOConnor/klipper/files/2634761/log-failed-sdcard.log.gz)

When I initiate print using printer menu it fails with the error "extruder temp below the minimum" about 10 minutes into print. My understanding it times out like it would normally do if I pre-heated but no job started for some time.

jakep82, 2018-11-30:
Have you verified that your gcode doesn't have a temperature change at a certain layer height?  It looks like the target temperature was changed from 200 to 0 in the middle of the print.

jakep82, 2018-11-30:
Actually looking closer, it seems the bed was shut off a few seconds before the hotend.  This is weird.

akpch, 2018-11-30:
No, there is no temperature change in t hat point. Exactly same gcode printed succesfully when the print was initiated via octoprint.

KevinOConnor, 2018-11-30:
It looks like something commanded Klipper to turn off the heaters.  From the log:
```
Stats 1132.3: gcodein=2515 mcu: mcu_awake=0.079 mcu_task_avg=0.000261 mcu_task_stddev=0.000251 bytes_write=1059728 bytes_read=230150 bytes_retransmit=46 bytes_invalid=0 send_seq=21258 receive_seq=21257 retransmit_seq=49 srtt=0.011 rttvar=0.003 rto=0.025 ready_bytes=14 stalled_bytes=0 freq=16012592 heater_bed: target=45 temp=45.1 pwm=0.117 sd_pos=540552 print_time=1125.545 buffer_time=2.123 print_stall=0 extruder: target=200 temp=200.1 pwm=0.415
Stats 1133.3: gcodein=2523 mcu: mcu_awake=0.079 mcu_task_avg=0.000261 mcu_task_stddev=0.000251 bytes_write=1061458 bytes_read=230400 bytes_retransmit=46 bytes_invalid=0 send_seq=21289 receive_seq=21288 retransmit_seq=49 srtt=0.011 rttvar=0.004 rto=0.025 ready_bytes=18 stalled_bytes=0 freq=16012594 heater_bed: target=0 temp=45.0 pwm=0.000 sd_pos=541844 print_time=1126.454 buffer_time=2.030 print_stall=0 extruder: target=200 temp=200.0 pwm=0.415
Stats 1134.3: gcodein=2528 mcu: mcu_awake=0.079 mcu_task_avg=0.000261 mcu_task_stddev=0.000251 bytes_write=1062922 bytes_read=230644 bytes_retransmit=46 bytes_invalid=0 send_seq=21315 receive_seq=21315 retransmit_seq=49 srtt=0.011 rttvar=0.004 rto=0.028 ready_bytes=18 stalled_bytes=0 freq=16012596 heater_bed: target=0 temp=45.0 pwm=0.000 sd_pos=542954 print_time=1127.753 buffer_time=2.327 print_stall=0 extruder: target=200 temp=200.0 pwm=0.415
Stats 1135.3: gcodein=2528 mcu: mcu_awake=0.079 mcu_task_avg=0.000261 mcu_task_stddev=0.000251 bytes_write=1064554 bytes_read=230893 bytes_retransmit=46 bytes_invalid=0 send_seq=21344 receive_seq=21343 retransmit_seq=49 srtt=0.011 rttvar=0.003 rto=0.025 ready_bytes=20 stalled_bytes=0 freq=16012594 heater_bed: target=0 temp=44.9 pwm=0.000 sd_pos=544278 print_time=1128.825 buffer_time=2.397 print_stall=0 extruder: target=200 temp=200.0 pwm=0.415
Stats 1136.3: gcodein=2544 mcu: mcu_awake=0.079 mcu_task_avg=0.000261 mcu_task_stddev=0.000251 bytes_write=1066955 bytes_read=231193 bytes_retransmit=46 bytes_invalid=0 send_seq=21385 receive_seq=21384 retransmit_seq=49 srtt=0.012 rttvar=0.002 rto=0.025 ready_bytes=18 stalled_bytes=0 freq=16012596 heater_bed: target=0 temp=44.8 pwm=0.000 sd_pos=545540 print_time=1130.090 buffer_time=2.660 print_stall=0 extruder: target=0 temp=200.0 pwm=0.000
Stats 1137.3: gcodein=2544 mcu: mcu_awake=0.094 mcu_task_avg=0.000276 mcu_task_stddev=0.000259 bytes_write=1068408 bytes_read=231447 bytes_retransmit=46 bytes_invalid=0 send_seq=21411 receive_seq=21410 retransmit_seq=49 srtt=0.013 rttvar=0.001 rto=0.025 ready_bytes=18 stalled_bytes=0 freq=16012595 heater_bed: target=0 temp=45.1 pwm=0.000 sd_pos=545926 print_time=1130.899 buffer_time=2.466 print_stall=0 extruder: target=0 temp=200.0 pwm=0.000
```

The key info is the "target=0" data points - the bed is turned off and then the extruder is turned off 3 seconds later.  If one looks closely at the logs, they'll notice that the number of bytes read from OctoPrint (the gcodein parameter) typically increments 5 bytes every 2 seconds (likely due to M115 requests).  However, during both of the above events the counter increases by a greater number - this is highly indicative that a heater command was sent to Klipper at these times.

Perhaps some OctoPrint plugin decided the heaters were idle and turned them off?

-Kevin

jakep82, 2018-11-30:
Reading this feature request, it sounds like Octoprint doesn't do this by default.  Perhaps you have the HeaterTimeout plugin installed?

https://github.com/foosel/OctoPrint/issues/960
https://plugins.octoprint.org/plugins/heatertimeout/

akpch, 2018-11-30:
My heaters get disabled automatically after about 10 min if I leave them alone. But no extra plugins installed, in fact many standard plugins disabled. I was sure it is Klipper function.

Easy enough to test. I will stop octoprint service and start print via printer menu again.

jakep82, 2018-11-30:
Oh, it is Klipper.  It does turn off the heaters after 10 minutes by default.

https://github.com/KevinOConnor/klipper/blob/master/config/example-extras.cfg#L323

akpch, 2018-11-30:
Restarted whole thing, turned off Octoprint and initiated new print. 12 minutes into print and it is still printing - past the point of previous failure.
 
This time I started printing cold. Must have something to do with the fact I always pre-heat extruder and bed before starting a print.That must be initiating timeout which is never reset when printing from SD card. This is next test. Unless someone confirms it before :)

akpch, 2018-12-01:
I could not recreate the issue with or without octoprint running.

KevinOConnor, 2018-12-09:
I'm going to close this as it does not appear to be a Klipper issue.  As above, there is high evidence during the initial failure that external g-code commands were sent to Klipper requesting that the heaters be disabled.

-Kevin

closed 2018-12-09 by KevinOConnor