Klipper3d/klipper #948: Move exceeds maximum extrusion

MSergW, 2018-12-01:
Your printer's firmware reported an error. Due to that OctoPrint will disconnect. Reported error: Move exceeds maximum extrusion (1.671mm^2 vs 0.640mm^2)

[octoprint.log](https://github.com/KevinOConnor/klipper/files/2636249/octoprint.log)

jakep82, 2018-12-01:
This has been reported and discussed many times in the past.  This is likely your slicer doing something weird.

https://github.com/KevinOConnor/klipper/issues/876
https://github.com/KevinOConnor/klipper/issues/539
https://github.com/KevinOConnor/klipper/issues/461
https://github.com/KevinOConnor/klipper/issues/420
https://github.com/KevinOConnor/klipper/issues/146
https://github.com/KevinOConnor/klipper/issues/32

MSergW, 2018-12-02:
the first layer is printed normally, this error appears on the second layer

mzakelj, 2018-12-04:
I have also similar problem. It started after update of klipper. 

Send: N166973 G1 X69.909 Y69.539 E0.0125 F1890*1
Recv: ok
Send: N166974 G1 X69.924 Y69.344 E0.0474*98
Recv: ok
Send: N166975 G1 X69.960 Y69.192 E0.1799*107
Recv: // Move exceeds maximum extrusion (2.040mm^2 vs 0.640mm^2)
Recv: // See the 'max_extrude_cross_section' config option for details
Recv: !! Move exceeds maximum extrusion (2.040mm^2 vs 0.640mm^2)
Changing monitoring state from "Printing" to "Error: Move exceeds maximum extrusion (2.040mm^2 vs 0.640mm^2)"
Changing monitoring state from "Error: Move exceeds maximum extrusion (2.040mm^2 vs 0.640mm^2)" to "Offline (Error: Move exceeds maximum extrusion (2.040mm^2 vs 0.640mm^2))"
Connection closed, closing down monitor

What can it be ? Is sliccer ? I use S3D. Until the update it run perfect.

mzakelj, 2018-12-05:
I think i find a solution for S3D. Remove "sticky" parameters in Gcode section of the slicer. It will make bigger code, but printer will chew it just fine.

KevinOConnor, 2018-12-09:
I'm going to close this as we did not get a klipper log file.  @mzakelj - it's very hard to say what issue you are seeing without seeing the log from the incident.  If you think Klipper did the wrong thing, please open a new issue and attach the klipper log file.

-Kevin

closed 2018-12-09 by KevinOConnor

MSergW, 2019-01-24:
5 часов печати, оставалось всего 20 минут :(
Your printer's firmware reported an error. Due to that OctoPrint will disconnect. Reported error: Move exceeds maximum extrusion (1.232mm^2 vs 0.640mm^2)

MSergW, 2019-01-24:
[BULLSEYE_DUCT_9.27...gcode.zip](https://github.com/KevinOConnor/klipper/files/2793341/BULLSEYE_DUCT_9.27.gcode.zip)

MSergW, 2019-01-24:
еще такая же ошибка если делать юбку в 2 слоя высотой, на втором слое обязательно ругается на Move exceeds maximum extrusion

alfsoft, 2019-02-02:
> I think i find a solution for S3D. Remove "sticky" parameters in Gcode section of the slicer. It will make bigger code, but printer will chew it just fine.

This actually works, thanks. But instead I get constant stuttering when printer stops for several seconds midprint and then continues to print.

CopterFail, 2019-10-18:
The same error with cura, during a draft 0.3 layer print, at the beginning of the 2nd layer:

Fehler: Move exceeds maximum extrusion (1.200mm^2 vs 0.640mm^2)

Is there also a work around for CURA 4.2 known?

Hairyloon, 2020-05-20:
> 
> 
> This has been reported and discussed many times in the past. This is likely your slicer doing something weird.
> 
> #876
> #539
> #461
> #420
> #146
> #32

That's an outstandingly unhelpful comment: "Begone with your boring question, we have spoken of it before... didn't actually find a resolution, but it's more fun to not tell you that..."