Klipper3d/klipper #950: M163 mixing filaments : support for rxtruder with multiple filament :  for clay printing

nuhwandah, 2018-12-02:
I'm in the process of migrating from Marlin, and would like to use Klipper on my RepRap Clay printer that uses g-code M163 to divide extrusion between a ram and nozzle. I'm not too great with the tech side (it took me months to get my clay printer going in Marlin and I'd like to try using Kilpper.) I was wondering if anyone would be interested in the same thing ?

joris57, 2019-03-07:
+1 for mixing extruder

kakou-fr, 2019-03-17:
+1 for mixing extruder

KevinOConnor, 2019-04-01:
I'm going to mark this as inactive as it doesn't look like it is currently being worked on.

-Kevin

closed 2019-04-01 by KevinOConnor

bguse, 2019-09-23:
Hi Kevin, what do you need to get forward on this feature? :-)

hevilp, 2019-11-02:
Hi @KevinOConnor ,
is there any movement here?

Thanks

KevinOConnor, 2019-11-13:
An interested developer would need to implement and submit this.  It is not something that I personally will have time to work on in the near future.

-Kevin

KevinOConnor, 2022-02-10:
FYI, with the latest code (including #5210 ) it may be possible to configure extruder_stepper motors that can push multiple filament spools simultaneously.

-Kevin