Klipper3d/klipper #951: Start klipper with negative nice, so it has precedence over other processes

dothebart, 2018-12-02:
Modifying the init script like this adding `--nicelevel -10` will run klipper with negative nice value, giving it precedence over other processes, so timeouts etc. become less likely:
```
start)  log_daemon_msg "Starting klipper" $NAME
        start-stop-daemon --start --quiet --exec $KLIPPY_EXEC \
                          --background --pidfile $PIDFILE --make-pidfile \
                          --chuid $KLIPPY_USER --user $KLIPPY_USER \
                          --nicelevel -10 -- $KLIPPY_ARGS
        log_end_msg $?
```

KevinOConnor, 2018-12-03:
The Klipper host software has been designed to not require special priority.  If you encounter an issue, you should attach the klipper log file from the event so we can determine what the error is.  See: https://github.com/KevinOConnor/klipper/blob/master/docs/Contact.md

-Kevin

KevinOConnor, 2018-12-17:
If there is no further comments on this, I'm inclined to close this issue.

-Kevin

dothebart, 2018-12-17:
running with negative nice got me way further down with a big piece to print where there would be aborts before. Since the Raspi and its architecture to plug everything into USB tends to be I/O bound in many cases, I'd still see the nice level as a good way to tell the system what to do first. However, currently stabilizing my setup, I may give you more feedback in the next weeks.

KevinOConnor, 2019-01-25:
Okay, thanks.  I'm going to close this as I think the Klipper host software should be able to run without special privileges.  If there's some reason that it can not, then we should treat that as a bug and fix it in the software.

-Kevin

closed 2019-01-25 by KevinOConnor

dothebart, 2020-01-10:
`MCU shut down: Timer too close` on a different setup without nice: 
- trigorilla board 
- Anycubic i3 Mega
- Bananapi (Allwinner A20 CPU Dual core cortex A7) Armbian OS, no cam.
so I'll re-inforce my statement: nice can't hurt but fix short commings in helping the kernel to decide what to do first.

KevinOConnor, 2020-01-15:
My recommendation is to not use nice on Klipper nor OctoPrint.  Raising the priority of a process raises the priority of all the threads in that process.  This is wrong for both Klipper and OctoPrint (as both contain threads that should have lower priority).  The Linux kernel generally does a good job of identifying those threads that are "interactive" and gives them higher priority - using nice effectively disables those heuristics.

-Kevin