Klipper3d/klipper #953: Axes Movement with panel

xamxamxam, 2018-12-02:
Hello
I am using Klipper ina a corexy selfmade printer with ramps 1.4, in conf I have 300 as max limit for x and y axes.
when I move with Gcode commands all ok but when I move with panel menu the extruder stops at 200.
Any suggestions?

uuzi, 2018-12-02:
as far as I understand (and i'm by no means an expert, since i just installed klipper a few days ago) you'll have to change that in the menu config.

see /klipper/klippy/extras/display/menu.cfg

there you'll find something like:
```
[menu __control __move_10mm __axis_x]
type: input
name: "X:{0:05.1f} "
parameter: toolhead.xpos
input_min: 0
input_max: 200.0
input_step: 10.0
gcode: G1 X{0:.1f}
```

there input_max should be edited accordingly

KevinOConnor, 2018-12-09:
As mentioned, it may be necessary to update the menu configuration.  Otherwise, if that does not address the issue, you'll need to attach the klipper log file from the event - see: https://github.com/KevinOConnor/klipper/blob/master/docs/Contact.md

-Kevin

KevinOConnor, 2018-12-17:
I think this question was answered, so I'm going to close this issue.

closed 2018-12-17 by KevinOConnor