Klipper3d/klipper #954: TMC2130 on MKS SBase - SPI pins and settings

speendo, 2018-12-02:
Hello!

I have to apology, this isn't a real bug report. However, as far as I know there is no Klipper forum yet...

I try to get an MKS SBase to work with my new Watterott TMC2130 drivers. I also got me some of these expansion boards (https://www.aliexpress.com/item/3D-Printer-Accessories-A4988-DRV8825-External-Stepper-Motor-Driver-Board-Panel-Additional-Adapter-Expansion-Board-for/32804550389.html) to have a proper socket.

Although I think I did the wiring on the expansion boards correctly (voltages look right) I cannot get my stepper motors to move.

Maybe I didn't get the SPI pins right or my settings might be wrong...

Can somebody help me figure out the correct wiring for the TMC2130 on the MKS SBase or - if this is not the cause of the error - lead the way to the correct settings?

I will most gladly provide additional information if that helps!

Thank you in advance!

KevinOConnor, 2018-12-03:
In order for us to assist, you need to attach the klipper log file from the event - see: https://github.com/KevinOConnor/klipper/blob/master/docs/Contact.md

speendo, 2018-12-05:
Sorry for my late response! I had to dig deeper in what SPI actually means and do some tests with a RepRap Discount 12864 display that I actually got working.

So, here are my findings:

**What actually works**

    [tmc2130 stepper_x]
    cs_pin: P0.16
    microsteps: 32
    interpolate: True
    run_current: 0.516129032
    stealthchop_threshold: 100

[image]
hold_current: 0.133194589

    ...
    cs_pin: P0.28
    ...

[image]

**What doesn't work**

It would be advantageous to use EXP2 instead of EXP1, because EXP1 is already used by the display.

According to [this](https://github.com/makerbase-mks/MKS-SBASE/issues/25#issuecomment-354115961) it would be necessary to switch from SPI0 to SPI1 and configure it like this

    ...
    cs_pin: P0.28
    ...

[image]

However, I cannot seem to make this work.

Is SPI1 active?

If yes, do I somehow have to select SPI1?

An even better way would be to use the J8 pins (according to [this](https://github.com/MarlinFirmware/Marlin/issues/9863) they do it in Marlin 2.0, however, I don't know if this is doable without software SPI...

A second hardware SPI should be available though.

Please also find my `klippy.log` attached.
[klippy.log](https://github.com/KevinOConnor/klipper/files/2650516/klippy.log)

KevinOConnor, 2018-12-09:
Currently, the Klipper lpc176x micro-controller code only supports spi bus 0.  It is possible to do hardware spi on the lpc176x using spi pins p0.7, p0.8, and p0.9, however that would require a code change (to src/lpc176x/spi.c) to enable support for both spi bus 0 and bus 1.

FWIW, it should be possible to control the tmc2130 chips via spi0 and then use any three other pins to control the st7920 display.  (The st7920 does not actually use spi - it just needs three general purpose pins.

-Kevin

speendo, 2018-12-13:
Thank you!

I have to say, I really like Klipper a lot because of this positive attitude you have!

I will soon check if I have enough free pins to connect the display and give this a try!

Thanks!

speendo, 2018-12-18:
update: tried to connect the Display on other pins: works.

The screen had hardly any contrast, but this is probably just an issue of correctly connecting the power pins. I will report back once I have a working solution.

gorbi87, 2018-12-18:
Hi,
 
i have a Problem with TMC2130 and SPI. I wire the TMC like Picture above on EXP1(CS on EXP2) and Connect the Extruder Motor(Just for testing). The Stepper Motor moves but SPI seems not working. 
Here my Config:

[extruder]
step_pin: P2.3
dir_pin: P0.22
enable_pin: !P0.21
step_distance: .000384
nozzle_diameter: 0.400
filament_diameter: 1.750
heater_pin: P2.7
sensor_type: ATC Semitec 104GT-2
sensor_pin: P0.23
min_temp: 0
max_temp: 280
min_extrude_temp: 15
pressure_advance: 0.030

[tmc2130 extruder]
cs_pin:P0.28
microsteps: 32
interpolate: true
run_current: 0.6
stealthchop_threshold: 0
sense_resistor: 0.110

driver_SGT: 0

When i send DUMP_TMC stepper=extruder.
I get :

 DUMP_TMC stepper=extruder
Recv: // GCONF:          00000000
Recv: // GSTAT:          00000000
Recv: // IOIN:           00000000
Recv: // TSTEP:          00000000
Recv: // XDIRECT:        00000000
Recv: // MSCNT:          00000000
Recv: // MSCURACT:       00000000
Recv: // CHOPCONF:       00000000
Recv: // DRV_STATUS:     00000000
Recv: // PWM_SCALE:      00000000
Recv: // LOST_STEPS:     00000000
Recv: ok

Do I have to specify SPI Pins in config or just connect to EXP1 as i did?

Any Ideas?

Thanks
[klippy.log](https://github.com/KevinOConnor/klipper/files/2691264/klippy.log)

speendo, 2018-12-20:
You certainly have to specify `cs_pin` (in your posting you are missing a blank on this setting by the way, don't know if that matters).

However, I wouldn't worry too much about the output of `DUMP_TMC`. I cannot test it right now, but as far as I remember, the output looked the same for me. However, as far as I see all the settings (including sensorless homing) are working...

gorbi87, 2018-12-21:
Everything works fine now incl Sensorless Homing. I can control the TMC2130 via SPI but still the same DUMP_TMC message. How can i use Stall detection and correction when the motor loose steps, like Home after Stall detect and continue the print? Is it already implemented yet?