Klipper3d/klipper #955: Can't get BLTouch probe working

J2xWo, 2018-12-02:
So i have been trying to get my BLTouch probe working using the [bltouch] module but no luck.
What am I doing wrong? X and Y are homing fine but when i want to home the z axis it gives me this error:
`BLTouch sensor test failed`
I thought that if i just set up the module everything would be configured right or did i misunderstood the use of the module?
[klippy.log](https://github.com/KevinOConnor/klipper/files/2637516/klippy.log)

BlackStump, 2018-12-03:
try
[bltouch]
sensor_pin = ar11
control_pin = ^ar18

BlackStump, 2018-12-03:
Was the BLTouch working prior or is this a new install of bltouch to klipper?
If it is a new install have you wired it correctly?
Do you have the right sensor / control pins named in the config?
when you first power your printer does the bltouch do its self test  ie pin down, pin up, pin down. pin up ?

J2xWo, 2018-12-03:
worked before, when the printer starts it does its selftest too and succeeds doing so, no errors

KevinOConnor, 2018-12-03:
Have you verified that you have the correct pins defined for the bltouch?  The error cited above occurs when the bltouch is unable to raise the signal line during a test.

Separately, I don't think the sensor line should have a pullup enabled for it.

-Kevin

J2xWo, 2018-12-03:
Yes I double checked. I got the MKS Gen L board. the servo pin should be D11.
I just now tried again after figuring out why i couldnt connect anymore (turns out the port randomly changed from /dev/ttyUSB0 to /dev/ttyUSB1. no idea when and why)
So when i try to home z then the servo deploys the probe but the z axis doesnt move and it gives me the previous stated error

KevinOConnor, 2018-12-03:
For the serial port issue - see: https://github.com/KevinOConnor/klipper/blob/master/docs/FAQ.md#when-the-micro-controller-restarts-the-device-changes-to-devttyusb1

The Klipper bltouch code is new.  It looks like we'll need to add some diagnostic commands to help figure out what is occurring here.

-Kevin

BlackStump, 2018-12-03:
is the Bltouch from antclabs  https://www.antclabs.com/       or a clone ?

J2xWo, 2018-12-04:
Thanks a bunch! Will be looking forward to it. I got a BLTouch clone in case that makes a difference. Is there a workaround in the mean time?

BlackStump, 2018-12-04:
I have a genuine antclab sbltouch and changed to the new bltouch module with no problems worked as it did before.
I only ask as there is plenty of info on the web with clones not working as expected. I am not saying that is the reason that yours is not working just conveying that my bltouch seems to work fine with the new module.

drbobbob, 2018-12-04:
I have a bltouch clone and it isn't working for me.  I think that i have my config correct you can find it on my fork https://github.com/drbobbob/klipper/blob/anetconfig/config/my-anet-a6.cfg

i know that my bltouch works because I have been using it with marlin.  One issue I have with my sensor is that for marlin I have set a rather long BLTOUCH_DELAY of 1 second.  It was not behaving well with lesser delays.  I couldn't figure out how to transfer that setting over as well.

When I try a homing sequence I get the "Endstop z still triggered after retract".  I attempted to modify the bltouch python to increase the delay like I have in my marlin config, but that didn't change anything.

Let me know if there are any tests that you would like me to run.

tvercruysse, 2018-12-04:
Im in the same boat. when using the servo + probe modules my bltouch clone works (atleast the homing works, mesh leveling + homing gives me a G-code already specified). when using the new bltouch module the printer homes to the set X and Y and then spits Endstop Z still trigered after retract. i'd love to help solve this issue so hit me up if i can be of assistance @KevinOConnor

case-tm77, 2018-12-04:
Same problem here.
i have installed a clone and since the switch to bltouch-extra he always reports "BLTouch sensor test failed".
I received an original Antaclabs today and will report if the problem still exists after the exchange.

kennethpangky, 2018-12-04:
I have a clone and the genuine BLTouch. The exact same commands works on the real BLTouch but gives a "BLTouch sensor test failed" on the clone.

Not sure why, but both worked on Marlin.

BlackStump, 2018-12-04:
attaching your log files before and after might help also quick search on bltouch clone problems is 
5v supply issues depending on controller used.
Some clones signal is inverted compared to original bltouch.
Wires are incorrectly placed in the connector and or not color coded correctly so need checking .
3pin
brown is GND
red is 5v
orange is control signal
2pin
black is GND
white is Z min

kennethpangky, 2018-12-04:
For my case the wiring should be correct - Got them working under Marlin before I flashed Klipper. Any commands I can run to troubleshoot maybe? Be gentle, I am a noob.

BlackStump, 2018-12-04:
Kevin has something in the works that adds some debug commands to the bltouch, well I know he has as I noticed he updated his bltouch branch.
I have tried the branch and the debug commands work as expect on a original bltouch, on a clone I don't know as I do not have.
hopefully he will chime in to give you some guidance.

drbobbob, 2018-12-04:
I am pretty sure that my wiring is correct,  as it was working with marlin and I didn't change my wiring.

I have done my tests on the master branch.  If you would like me to retest on another branch I can do that as well.

@tvercruysse can you send me your probe + servo config?  I would like to test that with my probe.  Perhaps that will give us some insights into how to better arrange the bltouch module.

KevinOConnor, 2018-12-04:
I've updated the bltouch code in the Klipper master branch.  For those that are seeing issues, please update to the latest version (see https://github.com/KevinOConnor/klipper/blob/master/docs/FAQ.md#how-do-i-upgrade-to-the-latest-software for details on upgrading) and then run the steps below.

To test the "control pin" run the g-code command `BLTOUCH_DEBUG COMMAND=pin_down` and verify the pin physically moves down.  Then run `BLTOUCH_DEBUG COMMAND=pin_up` and verify the pin moves up.  If it does not, then recheck the control pin wiring and config definition.

To test the "sensor pin" run `BLTOUCH_DEBUG COMMAND=pin_down` followed by `BLTOUCH_DEBUG COMMAND=touch_mode`.  Verify that `QUERY_PROBE` reports the probe as not triggered, then manually push the bltouch pin and verify that `QUERY_PROBE` reports the probe as triggered.  If the above does not work, then recheck the sensor pin wiring and sensor config definition.

If the above tests okay, but one still receives "Probe triggered prior to movement", "Endstop Z still triggered after retract", or "BLTouch sensor test failed" errors when actually probing, then try increasing the new pin_move_time config setting (see config/example-extras.cfg for the details).

Success or failure, please report your results here.

-Kevin

drbobbob, 2018-12-04:
Thanks for the support.  It will be a while before I am back at my printer to test these changes.

KevinOConnor, 2018-12-04:
@J2xWo - I had a typo in the last commit.  It should be fixed now (commit 76b57ba0).  Try pulling the latest code again and retry.

-Kevin

J2xWo, 2018-12-04:
@KevinOConnor That fixed it, at least now I can add config setting. Sadly i still get the error message that the sensor test failed. even after i bumped it up to 1.2s. As before the probe gets deployed and then and then I get the error message :/

KevinOConnor, 2018-12-04:
@J2xWo - Okay, that's surprising.  Could you make sure nothing is near the probe and then run the following commands:
```
BLTOUCH_DEBUG COMMAND=pin_down
G4 P1000
QUERY_PROBE
BLTOUCH_DEBUG COMMAND=pin_up
G4 P1000
QUERY_PROBE
BLTOUCH_DEBUG COMMAND=touch_mode
G4 P1000
QUERY_PROBE
BLTOUCH_DEBUG COMMAND=reset
G4 P1000
QUERY_PROBE
```
After the above is complete, run an `M112` and then attach the klipper log file here.

-Kevin

case-tm77, 2018-12-04:
hey, i pulled the new branch and nothing changed for the clone.
but the original antaclabs bltouch works fine, without any changes (firmware and wires).

i made come tests, the klippy_original.log with the antaclaps and the klippy_orignal_clone.log with the clone at the end.

the clone has a problem with the touch_mode (remains open).

thank you, all for your awesome work ;)
[klippy_original.log](https://github.com/KevinOConnor/klipper/files/2645892/klippy_original.log)
[klippy_original_clone.log](https://github.com/KevinOConnor/klipper/files/2645893/klippy_original_clone.log)

jakep82, 2018-12-04:
Try flipping the sensor plug (2 pin) on the clone.  It must be connected with the signal and ground in the appropriate orientation.  Perhaps the clone is reversed.

KevinOConnor, 2018-12-04:
@case-tm77 - Alas, the log only grabs the last ~20 seconds of messages, so it didn't capture all the necessary info.  Could use try again with the clone using the attached g-code script?  After it completes, quickly run M112, then attach the klipper log here.

-Kevin

[test-bltouch.gcode.txt](https://github.com/KevinOConnor/klipper/files/2646233/test-bltouch.gcode.txt)

case-tm77, 2018-12-04:
@Kevin
I'll be happy to try it tomorrow.

drbobbob, 2018-12-05:
Turns out I needed to invert my input pin.  Everything works once I changed the sensor pin to not have an invert.  Thanks for adding the debugging commands and thanks for making me think of think of inverting the input.

Just wondering, does klipper have an equivalent to Marlin's M48 for probe accuracy?  I was chasing the accuracy of my probing there and I was wondering if I could give a test here.  I do know python coding so I might be able to lend a hand coding that up.

http://marlinfw.org/docs/gcode/M048.html

case-tm77, 2018-12-05:
so i made some quick test's but for the clone no change.

the clone remains always "open" or if the pin is market negative always "triggered."
i inverted the pin twice.
-first test normal
-second test with inverted signal pin
-third test with negative input pin in config and inverted signal pin
and last test with negative input pin in config and normal pin.

i hope the log can help.
[klippy_clone_test.log](https://github.com/KevinOConnor/klipper/files/2647376/klippy_clone_test.log)

J2xWo, 2018-12-05:
Sure here you go, hope it helps

[klippy.log](https://github.com/KevinOConnor/klipper/files/2647461/klippy.log)

kennethpangky, 2018-12-05:
> Turns out I needed to invert my input pin. Everything works once I changed the sensor pin to not have an invert. Thanks for adding the debugging commands and thanks for making me think of think of [...]

Which input pin are you talking about?

nandinho84, 2018-12-05:
I've been experimenting some issues configuring the BL Touch as well. I think now it works ok.
However, when probing the same spot ( I chose 3 samples) I got a very different results. In some cases I've got a diference of almost 0.2mm in the same spot (in some cases even more). When I arrive home, I can put here the probe results for the quad_gantry_level. However, I was hopping that someone could verify if this is happening to them as well.... Maybe not a klipper issue, but If anyone could point me what am I'm doing wrong that would be much appreciated

drbobbob, 2018-12-05:
I also was fighting with my BL Touch consistency when running Marlin.

Changes that I messed with:
1. Probe speed (slower is better)
2. Decrease height retraction between probes
3. Increasing wait time for BLTouch to improve stability of probing
4. Rewiring Z motors in series rather than parallel.  I was seeing that early probes were good but they drifted up over repeated probe sequences.  I thought this might be related one of my Z motors losing steps, series should keep them in sync better.

I originally had wide variance but I have gotten it down to under 0.05 mm 

What sequence of commands were you using to do the repeated probing?  Try running a larger amount of probes.  I would notice that over the course of ~20 probes I would get into a state that my probe heights would continue to increase, that led to to question the accuracy of my Z axis.

nandinho84, 2018-12-05:
@jakep82 I'm using a clone. However, I've already used this 3d Touch in other 3d printer with very reasonable results. In this printer, the biggest difference is the klipper itself and the fact that the wires have almost 2.5 meters lenght. So, or some klipper configuration or some interference from the longer wires (but they have to be this lenght)

KevinOConnor, 2018-12-05:
@case-tm77 - Alas, the test didn't have the right timing.  You could try again with the script below.  However, I suspect you might just need to increase your bltouch pin_move_time config setting.

@J2xWo - In order for the log to be useful, you'll need to run M112 within a couple of seconds of the test finishing.  Also, your log appeared corrupted - make sure to scp the log and attach the full contents.  You could try again with the updated g-code script below.

-Kevin

[test2-bltouch.gcode.txt](https://github.com/KevinOConnor/klipper/files/2649871/test2-bltouch.gcode.txt)

nandinho84, 2018-12-05:
@drbobbob I'm not using any commands other than the quad_gantry_level with 3 samples (it probes 4 places on the bed, 3 times each...). I have the height retraction at 10mm and the speed at 5mm/s.

drbobbob, 2018-12-05:
@nandinho84  I am pretty new to klipper firmware so I didn't know that command existed.  I had been doing all of my repeatability testing with marlin firmware because I hadn't gotten my octoprint setup done.  Marlin has a nice M48 command that tests for probe repeatability.  I'll see if I can figure out a way to get this into klipper.  Knowing that the leveling commands do multiple samples should make it easier to write that command.  Are you at the latest master?  Have you tried changing the pin_move_time config to give the probe time to settle?  Is it possible you fouled the probe when moving it between machines.  I have read that BLTouch probes are sensitive to getting "stuff" on the probe which interferes with smooth movement.  It is also possible that you are getting noisy results because the BLTouch spikes current pretty hard and may be resetting due to voltage sag.  2.5 meter long power leads would have increased inductance.  This could probably be mitigated by putting a capacitor near the bltouch itself.  Depending on your level of electrical knowledge this may be easy or hard.

kennethpangky, 2018-12-05:
I'm using a clone, the commands work (pin up/down/test/reset) but when it reaches G28 it says sensor test failed. I'm using the test2-bltouch.gcode provided. Attached is my klippy log.

[klippy.log](https://github.com/KevinOConnor/klipper/files/2650343/klippy.log)

jakep82, 2018-12-05:
@kennethpangky Have you tried removing the pullup from the sensor pin (PC4 instead of ^PC4)?

KevinOConnor, 2018-12-05:
@kennethpangky - in order for the log to help an M112 command must be run right after the gcode script completes.  Though, I suspect you just need to increase the bltouch pin_move_time config setting.

-Kevin

BlackStump, 2018-12-05:
@drbobbob  you can create your own macro for testing probe repeatability
ie
[gcode_macro blt_test]
gcode:
  G28
  G1 Z10 F6000
  PROBE 
  G1 Z10 F1000 
  PROBE
  G1 Z10 F1000 
  PROBE
  G1 Z10 F1000

jakep82, 2018-12-06:
I did some probe testing tonight with a genuine BLTouch.  The single largest variance I had was .0375mm, but nearly every other probe was +/- .0125mm.

tvercruysse, 2018-12-07:
@J2xWo sorry for the late reply, if you are still in the need of a working setup. i got mine working, board is the ender 3 stock board (atmega 1284p) only thing you need to adjust is the probes z-offset to whatever yours is.

[printer.zip](https://github.com/KevinOConnor/klipper/files/2655381/printer.zip)

KevinOConnor, 2018-12-07:
Just to summarize, I don't know why some bltouch clones do not work with the new bltouch module.  The logs so far haven't been contained sufficient detail to track down what's different with these clones.  I'm attaching here an updated script for those that are having issues - one can try running this script and then attach the resulting klippy.log file.  Separately, a log from a session with pin_move_time=1.0 may also help.

-Kevin

[test3-bltouch.gcode.txt](https://github.com/KevinOConnor/klipper/files/2658870/test3-bltouch.gcode.txt)

J2xWo, 2018-12-08:
Sorry for being away! Uni kept me busy
Here is the log after running the gcode.  I noticed that it only deploys the probe but doesnt retract it.
I wanted to run the gcode again with a move time of 1s but for what ever reason it didnt wrote into the log even after restarting octoprint. Will upload it later then.

[klippy - move time .2s.log](https://github.com/KevinOConnor/klipper/files/2659477/klippy.-.move.time.2s.log)

case-tm77, 2018-12-08:
so i have done a new test with my bltouch-clone with the test3-bltouch.gcode file
but when i try to home Z klipper says "BLTouch sensor test failed"

[klippy_clone_test3.log](https://github.com/KevinOConnor/klipper/files/2659622/klippy_clone_test3.log)

fredizzimo, 2018-12-08:
It seems like if you start the printer with the bltouch pin down, then it won't pull the pin up before homing and it will fail. If the printer is started with the pin up, then everything seems to work fine. At least that's how it behaves for me.

You can easily verify this:
```
BLTOUCH_DEBUG COMMAND=pin_down
RESTART
G28
```
The probing will fail with "Recv: !! BLTouch sensor test failed"

If you now run this
```
BLTOUCH_DEBUG COMMAND=pin_up
G28
```

Then the homing will succeed.

So the fix should be as easy as adding a `BLTOUCH_DEBUG COMMAND=pin_up` command before the homing process starts. And this should be done before homing x and y, since it could be fatal if it tries to move with the pin down. 

I changed the code in to `test_sensor`
```
self.send_cmd(print_time, 'reset')
self.send_cmd(print_time + self.pin_move_time, 'pin_up')
home_time = print_time + 2 * self.pin_move_time
```
And it seems to work, but it's not a good sollution, since x and y is homed first, so it needs to be fixed in some other way.

fredizzimo, 2018-12-08:
It also fails to do the self-test when restarting the printer with `RESTART`, which is part of the reason, since the self-test moves the pin to the up state.

Also it seems like we could add more sanity checks. I will take a look at these issues and also see if I can do something about [#387 - BLTouch uses trigger test for actually probing](#387). I hope I will have a pull request ready later today.

oleplay, 2018-12-08:
@fredizzimo This doesn't fix it for me.
I still get `BLTouch sensor test failed` after the pin was pushed down while homing.

fredizzimo, 2018-12-08:
@oleplay, strange, did you restart the klipper service after making the changes? If you did, then we are most likely dealing with different issues.

Also, without the "fix", it works for me, as long as the pin is in the up state when staring the homing, if it's down, then it fails.

oleplay, 2018-12-08:
@fredizzimo I did not make any changes yet. I just tried homing with the pin up and pin down, both fail for me.

oleplay, 2018-12-08:
@fredizzimo I just tried changing the code and restarting the service. No change still fails to Home.

fredizzimo, 2018-12-08:
@oleplay, It's probably the different behaviour between the clone and the original. I have the original one.

I checked the marlin source code and it does the equivalent to this in order to reset the probe, if it shows that it's triggered at the start
```
BLTOUCH_DEBUG COMMAND=reset
G4 P375
BLTOUCH_DEBUG COMMAND=pin_down
G4 P375
BLTOUCH_DEBUG COMMAND=pin_up
G4 P1500 
```

After this the `QUERY_PROBE` command should show that the probe is not triggered. And hopefully homing should also work.

Could somone with a clone test if that works?

oleplay, 2018-12-08:
@fredizzimo I just tested your commands with my clone.
The `QUERY_PROBE` says `probe: open` but homing still doesn't work `BLTouch sensor test failed`

fredizzimo, 2018-12-08:
Ok, good. The current code does things in a differerent way, so there's probably something that causes it to enter the error condition again. But hopefully it will work after my changes, I will try to make it work exactly as Marlin.

fredizzimo, 2018-12-08:
I suspect that it's the touch_mode, that's not working on the clones. Could you run the following macro and paste the terminal output

```
[gcode_macro touchtest]
gcode:
    BLTOUCH_DEBUG COMMAND=reset
    BLTOUCH_DEBUG COMMAND=pin_down
    BLTOUCH_DEBUG COMMAND=pin_up
    G4 P1500
    M400
    QUERY_PROBE
    QUERY_ENDSTOPS
    BLTOUCH_DEBUG COMMAND=touch_mode
    G4 P1500
    M400
    QUERY_PROBE
    QUERY_ENDSTOPS
```` 

For me it shows this
```
Send: touchtest
Recv: // Sending BLTOUCH_DEBUG COMMAND=reset
Recv: // Sending BLTOUCH_DEBUG COMMAND=pin_down
Recv: // Sending BLTOUCH_DEBUG COMMAND=pin_up
[...]
Recv: // probe: open
Recv: x:open y:open z:open
Recv: // Sending BLTOUCH_DEBUG COMMAND=touch_mode
Recv: // probe: TRIGGERED
Recv: x:open y:open z:TRIGGERED
Recv: ok
```

The original code assumes that the endstop is triggered after the reset, which is why it failed for me with the probe in the down position. On the clones, the touch mode might not be implemented at all, therefore never reports the triggered state under those conditions.

oleplay, 2018-12-08:
Sure my Output:

Send: touchtest
Recv: // Sending BLTOUCH_DEBUG COMMAND=reset
Recv: // Sending BLTOUCH_DEBUG COMMAND=pin_down
Recv: // Sending BLTOUCH_DEBUG COMMAND=pin_up
Send: M105
Recv: // probe: open
Recv: x:open y:open z:open
Recv: // Sending BLTOUCH_DEBUG COMMAND=touch_mode
Send: M105
Recv: // probe: open
Recv: x:open y:open z:open

fredizzimo, 2018-12-08:
Thanks, that confirms that the touch mode is not working. If the pin is in the up state at the end of the test, it should return `TRIGGERED`, but for you it doesn't.

Hopefully this will work better when switching to the 10 degree probe. My progress with that is a bit slow, it takes some time to figure out the inner workings of Klipper.

oleplay, 2018-12-08:
I found this issue from Marlin
[https://github.com/MarlinFirmware/Marlin/issues/5817#issuecomment-279312462](url)
Maybe that helps

KevinOConnor, 2018-12-09:
@case-tm77 , @J2xWo , @fredizzimo - if you run a `BLTOUCH_DEBUG COMMAND=pin_down` then `BLTOUCH_DEBUG COMMAND=touch_mode` and then manually push the pin up a small amount, what does `QUERY_PROBE` report?

Also, I have a further refined test - if you have time, can you run the script below and forward the log?

-Kevin

[test4-bltouch.gcode.txt](https://github.com/KevinOConnor/klipper/files/2660716/test4-bltouch.gcode.txt)

fredizzimo, 2018-12-09:
@KevinOConnor 
One a genuine bltouch it reports triggered as it should. One a clone, I believe it still reports open. I don't think the touch mode is implemented on the clones at all.

I'm still working on my changes, which implements the proper 10 degree mode, and takes advantage of the built in error checking that the BLTouch has.

I will also enable an option, which enables touch mode error checking, similar to what you already have (but only for error checking), which can be enabled on genuine BLTouch probes, but not on clones. The advantage with that, is that it can detect the situation where the endstop is not configured with a pull up resistor. 

I has taken longer than I would have hoped, partly because of me being unfamilar with the klipper code, and partly because the BLTouch is not well documented. For example the error checking seems to take quite long to trigger, I think close to 1 second after you try to set the pin down, and it hits the bed prematurely. 

One small not about these, changes, they will not work without having a small z-hop after homing, otherwise you could never home if the nozzle is too close to the bed when you start.

I would also like to add the bltouch checks before any homing move on any axis is taken place. Since if you are in an error condition, the pin might be ouching the bed, and if you move to the side at that point then it could destroy the probe.

[153 more replies]

closed 2019-01-25 by KevinOConnor