Klipper3d/klipper #956: Delta Calibration, probing points isn't a circle

Hywelmartin, 2018-12-03:
Hello all..

When I issue a delta calibration, it doesn't probe in a circle with a center at x0,y0.. the first point is at x0,y0 but then it probes at the C tower(really close, bending belts) but at the other towers A, B it's ~2cm from the belts. If I decrease the probable area it's the same but a little smaller circle( no crashing, bending belts)..

I have cleared the config and even added x and y offsets( set to 0)
I can print and move around the whole bed and it stops with similar distance left to the edge..

I tried to visualize  it with some filament this time with smaller radius
[image]

I'm clueless 

```
[probe]
pin: ^!ar18
# Probe detection pin. This parameter must be provided.
x_offset: 0.0
#   The distance (in mm) between the probe and the nozzle along the
#   x-axis. The default is 0.
y_offset: 0.0
#   The distance (in mm) between the probe and the nozzle along the
#   y-axis. The default is 0.speed: 10.0
z_offset: -00.1200
#   The Z height (in mm) of the head when the probe triggers.
speed:15

[delta_calibrate]
radius: 78
```

```
DELTA_CALIBRATE

probe at 0.007,-0.008 is z=-0.083220  [OK]

probe at -0.006,74.100 is z=0.026066 [0,78..  why 74]

probe at -60.798,35.100 is z=-0.089328

probe at -57.419,-33.152 is z=-0.078696

probe at 0.008,-54.596 is z=0.066705 [0,-78]

probe at 50.664,-29.248 is z=-0.005567

probe at 54.032,31.200 is z=-0.216518
```

[klippy.logg.zip](https://github.com/KevinOConnor/klipper/files/2640940/klippy.logg.zip)

KevinOConnor, 2018-12-09:
Interesting.  That looks like the XY dimensions are way off.  Could you run the enhanced delta calibration (described at: https://github.com/KevinOConnor/klipper/blob/master/docs/Delta_Calibrate.md#enhanced-delta-calibration ) again?  If possible, attach the klipper log file that includes the calibration object print and the post calibration (DELTA_ANALYZE) steps.

-Kevin

Hywelmartin, 2018-12-09:
I don't have that log anymore(if its not in the attached one) but I have the measurements,and they are not that bad... the worse was 75.99 and the rest was 75 plus/minus ~0.2
I will do another one(starting from scratch) soon.. this is mostly annoying..

sidddy, 2018-12-11:
I also experienced this behaviour on my delta and checked the coding in delta_calibrate.py: it looks like this weird pattern is a feature, not a bug:

```
    # Calculate default probing points
        radius = config.getfloat('radius', above=0.)
        points = [(0., 0.)]
        scatter = [.95, .90, .85, .70, .75, .80]
        for i in range(6):
            r = math.radians(90. + 60. * i)
            dist = radius * scatter[i]
            points.append((math.cos(r) * dist, math.sin(r) * dist))
```
The `scatter` array determines the distance for each individual probe point, so your observed pattern seems to be expected...

KevinOConnor, 2018-12-11:
Oh, is that what the original question was?  If so, then yes!  Delta_calibrate will do that.  It's intentional and required for good calibration.  In order to get enough information to calibrate we need to choose positions on the bed that are not mirror images of each other - so the points need to be scattered a little.  Choose a radius parameter such that it is valid to probe at any point when the nozzle is within that radius.

-Kevin

Hywelmartin, 2018-12-12:
then it was good that I didn't had any time to clean out the config and start over...

Feel free to use my pictures in a guide if you want

Thanks anyhow /Martin

closed 2018-12-13 by Hywelmartin