Klipper3d/klipper #957: Z offset in tune menu

AndyWo83, 2018-12-03:
Hello,
that with the z-offset in the tune menu doesn't work.
Do I have to adjust something in the printer.cfg here so that works or in another *.cfg?
I entered a larger z-offset during printing and was unable to detect any reaction on the spindle because the menu.cfg says "realtime: true".
I hope someone can help me...

mcmatrix, 2018-12-04:
In order to assist, you need to attach the klipper log file - see: https://github.com/KevinOConnor/klipper/blob/master/docs/Contact.md

AndyWo83, 2018-12-04:
Hello,
just tested it again and now it works.
Yesterday I had tried it in three different print jobs and it wasn't there.
The issue can be closed...

closed 2018-12-05 by AndyWo83