Klipper3d/klipper #958: ENDER 3 + BL touch with EZout board

banolka, 2018-12-04:
Hi,

I'm trying to setup Ender 3 with BL touch support connected to LCD cable on original Creality 1.1.3 board with EZOUt LCD pin 27 adapter). How should I modify Ender 3 example of klipper to correctly address usage of BL touch as mesh bed leveling?

kennethpangky, 2018-12-04:
These are the parts I added/edited.

[bltouch]
sensor_pin: ^PC4
control_pin: PA4
x_offset: -41
y_offset: -5
z_offset: 0.3
speed: 5.0

[bed_mesh]
speed: 50
horizontal_move_z: 5
samples: 1
sample_retract_dist: 2.0
min_point: 50,30
max_point: 220,220
probe_count: 3,3

[homing_override]
set_position_z: 5
axes: z
gcode:
 G90 ; Uncomment these 2 lines to blindly lift the Z 2mm at start
 G1 Z7 F600 ; Move Z up a bit to prevent scratching
 G28 X0 Y0
 G1 X120 Y120 F3600
 G28 Z0

[gcode_macro m205]
gcode:

[gcode_macro G29]
gcode:
 G1 Z10 F600
; SET_SERVO SERVO=bltouch ANGLE=120
 G4 P200
 G1 Z10 F600
 BED_MESH_CALIBRATE
; SET_SERVO SERVO=bltouch ANGLE=10
 G4 P100

banolka, 2018-12-04:
great so this is using the BL touch with pin27 cable on ender3? I will maybe try tonight :)

kennethpangky, 2018-12-04:
Yes, the same setup. Mine's a clone of the EZout board, but it should be the same.

banolka, 2018-12-04:
What is your endstop pin

kennethpangky, 2018-12-05:
There's a ender3 config in the examples folder. Everything is stock from there.

banolka, 2018-12-05:
More or less I got it working. What about the homing and mesh generation code? I used code like on marlin but it was not performing. Bltouch leg dopped and blocked table.

KevinOConnor, 2018-12-13:
It looks like the original question was answered so I'm going to close this issue.  If there are further questions, please be sure to attach the klipper log file - see: https://github.com/KevinOConnor/klipper/blob/master/docs/Contact.md

-Kevin

closed 2018-12-13 by KevinOConnor