Klipper3d/klipper #962: I2C support on Arduino Due

wesk-ms, 2018-12-05:
Hello,

I'm trying to configure Klipper to use an I2C oled display (ssd1306) connected to an Arduino Due MC.  In looking at the code, it doesn't appear that I2C support is enabled for the SAM3X8E platform.  Is that correct?  Just thought I'd ask before spending a lot of time reconfiguring mt printer.

Thanks!

KevinOConnor, 2018-12-07:
You are correct - the i2c support is not currently implemented on the sam3x8e processor.  It is believed that the src/sam4e8e/i2c.c code could work with the sam3x8e with some minor changes - but no one has done this yet.

-Kevin

wesk-ms, 2018-12-08:
Thanks for the verification.  If I get a chance to port that code over, I'll reach out and let you know.

Thanks again!

KevinOConnor, 2019-01-03:
FYI - the code at PR #1055 adds i2c support for the sam3x8c.  Though, you'll probably also need the code in PR #1048.

-Kevin

wesk-ms, 2019-01-04:
Thanks Kevin!  Appreciate the update.

KevinOConnor, 2019-01-08:
FYI - both of the PRs listed above have been merged into the master branch.  So, i2c support should now be working on the Due.

-Kevin

closed 2019-01-16 by KevinOConnor