Klipper3d/klipper #963: Consultant sought for programming

technicksdesign, 2018-12-06:


technicksdesign, 2018-12-06:
[image]
I’ve designed and built a machine that’s similar to a printer, but, only 2 axis, bunch of microswitches, TOF sensor for distance measuring..

We are based in Vancouver Canada.

I’m a little out of my depth on the programming side. Got some movement out of one stepper, think the PWM length too short for the controllers.. or the system is expecting feedback from other hard not present??

If your willing to assist, here’s the basics of the machine I made (see pics):

- There are two gantries (the square frames inside) that move down then up, the left side to collect product to process, the right side to receive the collected product and dip it in a tray of silicon, those have limit switches top and bottom of travel limits, they also have micro switches inside them that detect the product trays are in place
- On the bottom left there is another micro switch to detect the initial feed tray of product is in place (These three micros are just to prevent drop and lift cycles if those conditions not met
- The right gantry has TOF sensor which travels with the gantry and would measure the distance to the liquid during the dip cycle, and stop and reverse the travel back up at a certain distance measured

Hardware is:

- Pi3+
- Arduino 2560
- Ramps 1.4
- Touch screen intended for status feedback and initiating the two possible production cycles
- Stepper controllers are StepperOnline DM542T to Nema23
- Multiple power supplies, 5, 12 and 24; 24 dedicated to steppers
- Adafruit VL53L0X Time-of-Flight Distance Sensor

Current status:

Pi loaded with OctoPi, MCU flashed, Octoprint server running operational and seeing the MCU, shows:

Send: M105Recv: ok T:0Send: 
status 
Recv: // Klipper state: Read

Just the motors and travel limit switches are connected to the Ramps board the moment.

So yeah, made it this far, but need an expert / highly knowledgeable person to help (Paid of course) with things like:

- Custom setting in the MCU flash config for pulse length, things specific to the DM542T
- Printer config file, we don’t have extruders etc, how to write / configure that properly so it doesn’t throw errors for missing hardware
- Programming the 2 possible production cycles, each gantry moving down then up
- Some way to trigger those cycle from the touch screen
[image]

technicksdesign, 2018-12-06:
Update: motors now run with the homing button and calibrate against endstops, but this only started working when I removed homing_speed: from printer.cfg   But once both have homed, trying to send a manual move command from OctoPrint just results in a slight scream from the motors, kinda like it did before I removed the homing_speed entries.

technicksdesign, 2018-12-06:
[klippy.log](https://github.com/KevinOConnor/klipper/files/2651453/klippy.log)

technicksdesign, 2018-12-06:
After trying to move after homing:

Send: G1 X10 F6000
Recv: ok
Send: G90
Recv: ok
Recv: ok
Send: M105
Recv: ok T:0
Send: M112
Send: N3 M112*34
Send: N4 M104 T0 S0*37
Recv: // Klipper state: Shutdown
Send: N5 M140 S0*96
Recv: ok
Closing serial port due to emergency stop M112.
Connection closed, closing down monitor
Changing monitoring state from "Operational" to "Offline (Error: Closing serial port due to emergency stop M112.)"

technicksdesign, 2018-12-06:
[printer.cfg.txt](https://github.com/KevinOConnor/klipper/files/2651464/printer.cfg.txt)

technicksdesign, 2018-12-07:
System now running well. Thanks guys.

KevinOConnor, 2018-12-13:
Any further questions on this, or should we now close this issue?

-Kevin

technicksdesign, 2018-12-13:
[cid:image003.jpg@01D492DD.720BF140]Hi Kevin,

Yes; there is some help needed.

I got the machine to the stage of it working; installed TouchUI, that’s all good. Runs well, calibrated, its now just a regular cnc machine that needs code.

But; need to figure out a couple of things beyond my knowledge, want to use one of two TOF sensors, type VL6180X and same without the X. They are already mounted and ready (see pics, mounted of the right side gantry).

Ref the pics.. there are two stages for this machine, the challenging bit being the side with the tray underneath, the plan there is that as it moves down, it dips some products in to silicone but only to a certain depth; the TOF sensor should kinda act as a ‘end stop’, measuring the distance from the gantry to the silicone as it moves down, slowly at that point, once a certain depth is achieved, stop and move up again. The reason I put the TOF there is that the silicone will be used up each dip (topped up as required) and the level will be changing, so it has to be measured actively each cycle.

The left side (where we feed in the products mounted on boards that slide thru the channels that project out each side) is fairly easy; will just move down to a designated point, stop, and move back up again to zero.

So here are the objectives in point form:

  *   Create a method to use the TOF depth sensor to actively measure the dip depth and cause the gantry to stop and move back to zero
  *   Create the gcode to run the cycles
  *   Modify the TouchUI interface, see pic.

Do you have time to help with this?

Kind Regards,

Nick Taylor.

technicksdesign, 2018-12-18:
Hi Kevin,

Just following up on this..

Could also use some advise about connecting the additional limit switches and the TOF range sensor.

Then work out how to program the system..

Kind Regards,

Nick Taylor.

________________________________
From: Nick Taylor <nicktaylorit@outlook.com>
Sent: Thursday, December 13, 2018 12:14:51 PM
To: KevinOConnor/klipper; KevinOConnor/klipper
Cc: Author
Subject: RE: [KevinOConnor/klipper] Consultant sought for programming (#963)

[cid:image003.jpg@01D492DD.720BF140]Hi Kevin,

Yes; there is some help needed.

I got the machine to the stage of it working; installed TouchUI, that’s all good. Runs well, calibrated, its now just a regular cnc machine that needs code.

But; need to figure out a couple of things beyond my knowledge, want to use one of two TOF sensors, type VL6180X and same without the X. They are already mounted and ready (see pics, mounted of the right side gantry).

Ref the pics.. there are two stages for this machine, the challenging bit being the side with the tray underneath, the plan there is that as it moves down, it dips some products in to silicone but only to a certain depth; the TOF sensor should kinda act as a ‘end stop’, measuring the distance from the gantry to the silicone as it moves down, slowly at that point, once a certain depth is achieved, stop and move up again. The reason I put the TOF there is that the silicone will be used up each dip (topped up as required) and the level will be changing, so it has to be measured actively each cycle.

The left side (where we feed in the products mounted on boards that slide thru the channels that project out each side) is fairly easy; will just move down to a designated point, stop, and move back up again to zero.

So here are the objectives in point form:

  *   Create a method to use the TOF depth sensor to actively measure the dip depth and cause the gantry to stop and move back to zero
  *   Create the gcode to run the cycles
  *   Modify the TouchUI interface, see pic.

Do you have time to help with this?

Kind Regards,

Nick Taylor.

KevinOConnor, 2019-01-06:
It seems you've made progress on this.  I'm going to close this github issue.  Let us know if you need further help.

-Kevin

closed 2019-01-06 by KevinOConnor