Klipper3d/klipper #964: Problem with heater after last update

nandinho84, 2018-12-06:
After the last update, I've noticed one problem in my MKS SBASE 1.3... The first time when I set the temperature, it rises as it should. However, If I bring the temperature down and then try to raise it again, then the temperature doesn't rise (Or maybe it raises, but the thermistor doesn't sense it).

Here is the log file:

[klippy.zip](https://github.com/KevinOConnor/klipper/files/2651605/klippy.zip)

jakep82, 2018-12-07:
From your log:
> Heater extruder not heating at expected rate

When you turn on a heater that is rapidly cooling, it's fairly easy to run into a heater verification error like this.  See the link for more information.  You can avoid this by increasing the check_gain_time.

https://github.com/KevinOConnor/klipper/blob/master/config/example-extras.cfg#L301

nandinho84, 2018-12-08:
[R@1n.txt](https://github.com/KevinOConnor/klipper/files/2659323/R.1n.txt)
You can see that the temperature is not stable at all... I've already run PID calibrationm, but the problem remains... I've already tried to control the heater with an external mosfet board, but the problem remains...

BlackStump, 2018-12-08:
have you tried a new heater?

jakep82, 2018-12-08:
It looks like either your heater has failed, or you have a wiring issue.

nandinho84, 2018-12-15:
I've just changed the heater and the thermistor and the problem remains. Then I've used a external mosfet and the problem not seems to change.
One thing I've notice though, When I try to set a temperature, the temperature keeps stable at that point, then when I try to print, the temperature goes crazy... See the attached log file and the print screen of the temperatures graphic (from -27min to -22min was the PID calibration, then from -17min to -7min was the manual temperature setting, then was the start of the print - it doesn't even started because it was taking ages to stabilize the temperature and then I hitted M112 to download the log file)
[klippy.zip](https://github.com/KevinOConnor/klipper/files/2682310/klippy.zip)
[image]

AxMod3DPrint, 2018-12-15:
As a note, the same thing happens with my Re-Arm and is down to the board
hardware itself. It was the same with Smoothie, even with the thermistors
coming from the Re-Arm and not the RAMPS, although it was less noticeable.
If you have an AVR based board, I would suggest wiring that up as a check,
as as soon as I put a RAMPS/Mega combo in the printer, the problem
disappeared. In some cases the LPC needs a lot more smoothing than an AVR
based board due to the signal noise.

I would also suggest making sure the wires to the board from the thermistor
are of high quality and a decent wire gauge to help this out.

nandinho84, 2018-12-15:
In that case, shouldn't it happen as well when trying to reach a certain temperature without printing? This happens only when I try to print something. Before that, if I try to set a temperature manually, it remains stable all the way...

AxMod3DPrint, 2018-12-15:
No, it's noise generated from other parts of the board like backfeeding
from the stepper drivers. As such, it's a pain in the arse to get rid of.

nandinho84, 2018-12-16:
But I didn't had this problem with smoothieware...
And the problem starts even with the steppers not active (during the heating up prior to the printing)...

nandinho84, 2018-12-16:
I don't have any spare board... I guess I'll have to order two mega + two ramps to test it...

rtbon, 2018-12-19:
Hello, same board (MKS SBASE), same problem, after PID tuning, temperature never stable up and down from 0.1 to 5°C sometimes more. I've already tried to play with the smooth_time but same trouble. Someones talk about a median filter… but I've find nothing ti this way in klipper… I've never had this problem with smoothieware… so I hope someone will find a fix because I prefer from away Klipper

nandinho84, 2018-12-19:
For me it seems some kind of bug... I have no knowledge to dig into it, but it seems strange to me. I thought I was the only one having this problem and I was wondering about downgrade to a 8bit board to fix the problem... But maybe there's some kind of bug. 
My temperatures are stable when not printing... Even if I try to level the gantry with the G32 or while homming with the G28 command, the temperatures are stable. However in the second that I select "print", the temperatures go crazy up and down. This happens to you as well?

KevinOConnor, 2018-12-21:
This issue looks like a dup of #842.  It's a known problem with the lpc176x chips (yes - worse noise during movement is exactly the symptoms of the poor adc behaviour on this chip).  Increasing the heater smooth_time is the best recommendation at this point.

-Kevin

nandinho84, 2018-12-22:
> This issue looks like a dup of #842. It's a known problem with the lpc176x chips (yes - worse noise during movement is exactly the symptoms of the poor adc behaviour on this chip). Increasing the heater smooth_time is the best recommendation at this point.
> 
> -Kevin

I don't notice any noise during movement. For instance, if I entry an G32 command to level the gantry (4 probe points, 3 samples each, three times this routine), the temperature remains stable. The problem only starts when I press the "print" button, even before the first movement of the steppers (during the initial heat up).

KevinOConnor, 2018-12-27:
Lets move this to #842.

-Kevin

closed 2018-12-27 by KevinOConnor

KevinOConnor, 2019-01-25:
FYI, it is believe that this fixed now - see issue #842 for the details.

-Kevin