Klipper3d/klipper #967: Position Control Implementation

michaelwu21, 2018-12-07:
Hi, I am looking to adapt klipper code such that it would allow the use of SPI communicated position signals on trinamic motion controllers, while still retaining the time accuracy of klipper.

There was an issue opened on this earlier but I have done some research since.

Basically, what I need to change:
1) Move.set_junction to calculate an s-ramp profile instead of trapezoid, using 4 bow(pps3) values
2) Instead of an iterative solver, it can quickly solve for the time of the entire motion(trinamic has formulas for this)
3) Send just the time, position, start/max/end velocities of the motion to the mcu, which is then added to the queue
4) ISR will just take next time + pos in queue, spi write new info to registers, and set next timer interrupt.

The motion controller should finish the motion exactly in the time calculated, and if stalled, the controller will dynamically adjust motor speed(dcstep) to arrive at target position in that time. The IC can also be programmed to trigger it's INTR pin if such a standstill condition occurred. Furthermore, the START pin can be used to automatically synchronize motors if dcstep(slowing down for more torque) was to occur on one of them. Finally, if the motion timing is still not accurate enough, it also has a built in motion buffer(shadow register) so future motions(up to 2 iirc) can be queued and executed once position target is reached.

Would appreciate it if you could give me some guidance on getting started, specifically on steps 3 and 4 as I believe those would require the most amount of change. Steps 1 and 2 should be relatively straightforward.

Thanks,
Michael

michaelwu21, 2018-12-09:
Yes, I've already read through that documentation, however, I need to not just change high level functionality, but completely change the motion buffer. A motion will need to skip the step iterative solver and the position + velocities need to go directly to the microcontroller. This is the part I'm not exactly sure how to do. I've already figured out how to change kinematics to use a third dimension of motion(actual jerk) and find all needed values. Changing mcu code to send over spi instead shouldn't be too difficult, I'm just looking for a suggestion on how I should change the motion queue and the communication for this new info that needs to be sent over (pos and vels instead of steps) 

..

KevinOConnor, 2018-12-09:
Alas, I don't have sufficient knowledge of these motion chips to provide guidance on what future code should look like.  I may be able to assist with specific questions about the current code, or if you can point to specific areas of the documentation that are unclear.

-Kevin

michaelwu21, 2018-12-10:
Thanks!

I have not yet dug into the motion queue, but looking through the documentation, it looks like the host sends movement information of "interval, count, and add" to the mcu which is then added to a motion queue array. Is this correct and if so, I don't seem to understand what 'add' does? Knowing this would greatly help clarify what modifications I need to make.

Michael

KevinOConnor, 2018-12-11:
For each step pulse generated from a `queue_step` command, the micro-controller effectively does:
`generate_step_pulse(); next_wake_time = last_wake_time + interval; interval += add;`

So, `interval` is effectively a velocity, and in physical terms, the `add` parameter could be thought of as an acceleration (a change in velocity).  But, be careful not to confuse it with constant acceleration (a non-zero add will result in an acceleration, but it is not constant acceleration).

I don't generally think of the `add` term in a physical sense though.  It's really just a mechanism that enables the host code to do a better job at reducing the number of `queue_step` commands when the velocity of the head is changing.  Internally, Klipper finds the ideal step times for each stepper and then compresses them down to a series of `queue_step` commands - when the velocity of the axis is changing, the `add` parameter improves this compression.

-Kevin

michaelwu21, 2018-12-20:
Okay I have a Klipper host set up on linux so I can play around a bit with the code.

I created a host module for the tmc with config options of jerk1-4, encoder res and type, and it checks to make sure the x + y drivers are tmc set to 256 microstepping. I'm having a bit of an issue with the SPI, the output looks like this:
spi_send oid=3 data='\xcfRST\x00'
spi_send oid=3 data='\x83\x00T\x00"'
spi_send oid=3 data='\x84\x84@\x00\r'
spi_send oid=3 data='\xa0\x00\x00\x00\x06'
spi_send oid=3 data='\x8a\x00\x00\x00\x00'
spi_send oid=3 data="\xad\x00\x00'\x10"
spi_send oid=3 data="\xae\x00\x00'\x10"
spi_send oid=3 data="\xaf\x00\x00'\x10"
spi_send oid=3 data="\xb0\x00\x00'\x10"

Not sure what's wrong with it as I copied the set_register function from the 2130, and when calling the function directly with the values I'm using, it produces the correct output.

Could you give me some insight in how toolhead.py works? This is my current understanding, ToolHead gets a new position, which is then added to the move queue. It then calculates junction(which is look ahead?) and then set_junction which is the trapezoid generator? Is set_junction used to determine ramping + cruise + ramping for x and y separately or just as the sum of the x and y velocity vectors? What I'm a bit confused on is how the velocities are determined so that x and y motors will arrive at the target location at the same time, is the iterative solver what synchronizes this? 

Regarding mcu side, would it be feasible to add spi commands into the movement queue so that at a specified interval, it will do an spi write/read? Or just have a completely separate queue just for that? It should work as long as the host can call an spi write with a specific interval.

I had a few more questions but I'll wait off on those until trinamic responds first to them.

Thanks

KevinOConnor, 2018-12-21:
> Not sure what's wrong with it as I copied the set_register function from the 2130, and when calling the function directly with the values I'm using, it produces the correct output.

I'm not sure either - did you get some kind of error?

> Could you give me some insight in how toolhead.py works? 

I tried my best at: https://github.com/KevinOConnor/klipper/blob/master/docs/Code_Overview.md#code-flow-of-a-move-command .  If you can point out which parts of the description are clear and which parts are unclear then I can try to expand.

All of the code in toolhead.py is done in cartesian coordinates, and it calculates speeds/accel relative to the print.  So, none of these calculations are done relative to the stepper motors.  The code determines where the head should be moving (in a cartesian coordinates), how long the move should take, and the velocity/accel of the move - it's up to the kinematics code to then convert that movement into the associated stepper pulses to make that desired move occur.

> Regarding mcu side, would it be feasible to add spi commands into the movement queue so that at a specified interval, it will do an spi write/read?

That's a question I'd have for you, as it's your implementation!

-Kevin

KevinOConnor, 2019-01-03:
> I'm not sure how klipper does makes that the extruder will match the time the motion will take.

The Klipper high level code (in toolhead.py) figures out when each move will start, the duration of the move, and when the move will end.  With that timing, it calculates the step times for the move - with each step being schedule between the given move start and move end times.  The extruder steps are therefore synchronized because they are explicitly timed.  So, for example, if we start a move at time 12:34:47.123500 that lasts one millisecond and requires 100 extruder steps every 10us then we'll schedule extruder steps at 12:34:47.123510, 12:34:47.123520, 12:34:47.123530, etc.  These steps are then naturally in sync with the x/y steps because they too are explicitly timed.

> Any recommendations on how to have the host skip the iterative solver right to this?

Alas, I don't have any recommendations.  I don't have sufficient knowledge on how the tmc drivers work or your particular goals to give any design guidance.  So, I can try and describe how the current code works, but I feel any advice on future code would be too shallow to offer any benefit.

-Kevin

michaelwu21, 2019-01-03:
Is the calculation of the move duration done in set_junction()? I already have the code needed to do move duration calculations with s-curve.

Am I understanding the code flow correctly? After Move.move() is called, the motion information is passed into move_fill() which fills a move structure in C.  toolhead.kin.move() goes onto the kinematics classes which runs rail.step_itersolve() which then goes into rail class in stepper.py which is stepper.step_itersolve, which is mcu_stepper.step_itersolve(). 
step_itersolve() calls itersolve_gen_steps which uses the information from the move struct. 

After that, it uses the kinematics functions to find step times. I'm unsure about what itersolve_set_stepcompress is doing and what it means by set_stepper_kinematics?

Someone on discord pointed out to me that cruwaller has already started work on this in his fork, in 51xx.py. He does not have position control implemented yet but he has moved the drivers into a separate folder with an init code, so that the driver becomes an object in the printer class. In stepper.py, he checks if step/dir is not defined and if so, will set mcu_stepper to the driver, in this case with the 51xx, all the itersolve functions will be mapped to the ones inside 51xx.py. I saw your mechaduino example with the mcu_virtual_stepper and think that cruwaller's approach might be worth considering. 

What I plan on doing is to modify the itersolve_gen_steps code so that instead of outputting multiple queue_step, it produces one with an additional parameter like "spi = 1, steps = int" so that the mcu can check that if this parameter is true when this sequence is called in the per-stepper queue, and send steps over spi instead. Count will be set to 1 and interval will be how long the motion will take. 

However, I would much prefer having something like schedule_digital_out for an spi transaction, although my C is probably not good enough to write that.

Anything to watch out for when attempting this?

KevinOConnor, 2019-01-03:
> Is the calculation of the move duration done in set_junction()?

Yes - the move time is accel_t + cruise_t + decel_t.

> I'm unsure about what itersolve_set_stepcompress is doing and what it means by set_stepper_kinematics?

The "C move structure" contains the move start and end time, start and end position, along with the "velocity trapezoid" (as described in docs/Kinematics.md).  So, in effect, it describes what the desired position of the toolhead is at any point in time of the given move.  Given a desired toolhead position, we can use the kinematics formulas to determine the desired stepper motor position.  So effectively, we have a math formula: `position = desired_stepper_position(time)`.  The code wants something slightly different though - it wants to know: `time = desired_step_time(position)`.  The itersolve_set_stepcompress() code "solves" the desired_stepper_position(time) formula to find a time given a step position.

I'm not sure what you're asking wrt set_stepper_kinematics.

> I saw your mechaduino example with the mcu_virtual_stepper and think that cruwaller's approach might be worth considering.

I agree that the config file and code layout should be updated to account for more advanced stepper drivers such as the mechaduino.

The mcu_virtual_stepper is a short-term "hack".  FWIW, my current high-level plan is to enhance the host code so that it can produce and transmit to the mcu a list of desired stepper positions at fixed time intervals.  So, for example, the mechaduino might effectively get a list of positions every 200us - something like: at time 12:34:47.123400 be at position 1234, at time 12:34:47.123600 position 1234, at 12:34:47.123800 position 1235, 12:34:47.124000 => 1236, etc.  This would be easier for the mechaduino micro-controller to process and should not be harder for the host to calculate.  It will require that the list of positions be "compressed" so that they can be efficiently transmitted - a scheme similar to the compression of step times should work.

> However, I would much prefer having something like schedule_digital_out for an spi transaction, although my C is probably not good enough to write that.
> Anything to watch out for when attempting this?

I don't want to rehash issue #896, but I do feel managing precisely timed spi messages will be challenging to implement.

-Kevin

michaelwu21, 2019-01-07:
Okay, I wrote a somewhat working s-curve set_junction code, by obtaining a 4 jerk values from the 4361 config and calculates accel, cruise, and decel time. However, the G28 command tries to move at 5mm/s for 75mm at max accel? I haven't looked into homing code so I'm not sure what it's doing, but at a certain point(a value too high or low, in this case the 5mm/s) and the s-curve will overshoot. I think I can add a function that will limit jerk or accel if it will overshoot. I'm getting an error from step_compress but I believe that's an issue with accel not matching up with the ramp timings.

> I don't want to rehash issue #896, but I do feel managing precisely timed spi messages will be challenging to implement.

In what way is it hard to make spi precisely timed? Since my goal is all axis using position control, wouldn't they all be delayed by the same 5 microsecond transaction time? Haven't done much with spi(or low level stuff really), so maybe this timing is not precise? If so, I can always preload the motion into a shadow register(on the IC), so that when the actual motion needs to occur, a pin is pulled high on all motion controllers and all shadow register motions being execution simultaneously

KevinOConnor, 2019-01-10:
> In what way is it hard to make spi precisely timed?

The main difficulties I'd worry about are:
1. Performing the spi transfer quickly and with precise timing in the mcu code (in particular, avr platforms lack dma to do this).
2. Handling moves that nominally only take a few micro-seconds.  (Slicers tend to do goofy things on arcs.)
3. Corruption of requests between mcu and motion controller due to noise.  (Should a bit get flipped on a move request - is there a chance the motor will move thousands of steps incorrectly.)
4. Will minor timing errors add.  (It's not a big deal if a particular move starts a few microseconds late due to message timing, but if timing errors accumulate then it potentially leads to a situation where one axis is out of sync with another axis.)

I don't know much about these chips, so can't say if any of the above are an issue.  These are just some of the challenges that I'm generally aware of.

-Kevin

KevinOConnor, 2019-01-25:
Any further comments on this?

-Kevin

closed 2019-02-03 by KevinOConnor