Klipper3d/klipper #970: TMC2130 stepper not moving

cooked, 2018-12-07:
Hi, 
I'm using Klipper on Raspberry Pi connected to a  custom board with the STM32F103 mcu and TMC2130 drivers.
The motors are not moving although the signals (EN, STEP) looks fine on the oscilloscope and the SPI communication is working.
Here the output of DUMP_TMC and the klippy log in the attachments.
Any ideas?

Stefano

```
Send: DUMP_TMC STEPPER=stepper_y
Recv: // GCONF: 38000000
Recv: // GSTAT: 38000000
Recv: // IOIN: 38110000
Recv: // TSTEP: 38000fff
Recv: // XDIRECT: 38000000
Recv: // MSCNT: 38000002
Recv: // MSCURACT: 38019401
Recv: // CHOPCONF: 38000000
Recv: // DRV_STATUS: 38e00000
Recv: // PWM_SCALE: 38000000
Recv: // LOST_STEPS: 38000000
Recv: ok
[...]
Send: DUMP_TMC STEPPER=stepper_x
Recv: // GCONF: 38000000
Recv: // GSTAT: 38000000
Recv: // IOIN: 38110000
Recv: // TSTEP: 38000fff
Recv: // XDIRECT: 38000000
Recv: // MSCNT: 38000000
Recv: // MSCURACT: 3800f200
Recv: // CHOPCONF: 38000000
Recv: // DRV_STATUS: 38e00000
Recv: // PWM_SCALE: 38000000
Recv: // LOST_STEPS: 38000000
Recv: ok
[...]
Send: DUMP_TMC STEPPER=stepper_z
Recv: // GCONF: 38000000
Recv: // GSTAT: 38000000
Recv: // IOIN: 38110000
Recv: // TSTEP: 38000fff
Recv: // XDIRECT: 38000000
Recv: // MSCNT: 38000001
Recv: // MSCURACT: 3801a500
Recv: // CHOPCONF: 38000000
Recv: // DRV_STATUS: 38e00000
Recv: // PWM_SCALE: 38000000
Recv: // LOST_STEPS: 38000000
Recv: ok
[...]
Send: DUMP_TMC STEPPER=extruder
Recv: // GCONF: 38000000
Recv: // GSTAT: 38000000
Recv: // IOIN: 38110000
Recv: // TSTEP: 38000fff
Recv: // XDIRECT: 38000000
Recv: // MSCNT: 38000003
Recv: // MSCURACT: 3800f701
Recv: // CHOPCONF: 38000000
Recv: // DRV_STATUS: 38800000
Recv: // PWM_SCALE: 38000000
Recv: // LOST_STEPS: 38000000
Recv: ok
```

[klippy.log](https://github.com/KevinOConnor/klipper/files/2658490/klippy.log)

cooked, 2018-12-08:
You're probably right. I was fooled by the "legit" look of the DUMP_TMC output.
Does anybody have a DUMP_TMC output of a working SPI?

yschroeder, 2018-12-09:
@cooked: The SPI is doing all kinds of weird things currently for the STM32F103. For legits dumps: The registers should not all start with 0x38, that is the answer of the byte before (see #908). Afterwards, the SPI command sent to the TMC2130 are broken (see #909) so the drivers are not cofigured, resulting in motors not moving at all.

KevinOConnor, 2018-12-27:
I'm going to close this - lets use #908 to track this.

-Kevin

closed 2018-12-27 by KevinOConnor