Klipper3d/klipper #971: Why any error makes octoprint disconect?

rmcbc, 2018-12-07:
First of all I want to thank you the developer for this excellent job.
And after saying all this good things,

1 - Is it possible to avoid that errors like syntax errors, and others, makes octoprint disconnect?
     You should only show in terminal the errors, this way we don't have to repeat connection and lose all changes made.

jakep82, 2018-12-07:
Please take a closer look at the installation instructions.  Specifically 

> Enter the Settings tab again and under "Serial Connection" change the "Serial Port" setting to "/tmp/printer". Navigate to the "Behavior" sub-tab and select the "Cancel any ongoing prints but stay connected to the printer" option. Click "Save".

https://github.com/KevinOConnor/klipper/blob/master/docs/Installation.md

rmcbc, 2018-12-07:
Great. Thank you,

closed 2018-12-12 by rmcbc