Klipper3d/klipper #972: The documentation is insuficient so I ask some questions here....

rmcbc, 2018-12-07:
1 - Using the octoprint terminal how can I change the z_offset from [probe] ? Because I use a inductive sensor that needs +-0.10mm adjust depending on temperature.

2 - Is there a way to configure pulse frequency to different drivers, in marlin, A4988 is 1us and DRV8825 2us, and so on.

Thanks again for the good work.

jakep82, 2018-12-07:
Take a look at SET_GCODE_OFFSET.  I believe it's the only runtime option for adjusting Z.

https://github.com/KevinOConnor/klipper/blob/master/docs/G-Codes.md

rmcbc, 2018-12-07:
I try SET_GCODE_OFFSET but doesn't seem to take effect.
Or I'm doing something wrong because the only way to make it work properly is to one printer.cfg and change the z_offset.
I was trying to do place the G-Code in slicer:

G28
BED_MESH_CALIBRATE
SET_GCODE_OFFSET Z=3.15

BlackStump, 2018-12-07:
seems to be some confusion I think unless I am not understanding either but SET_GCODE_OFFSET is not a change to PROBE offset.
in your G-code added to you slicer I would expect you nozzle height to bed to be increased by 3.15mm not good for first layer I would suspect.
if your Probe Z offset is incorrect it will be nozzle to bed clearance which you can change by the runtime SET_GCODE_OFFSET Z  and in you example you want a change of 0.10
ie
G28
BED_MESH_CALIBRATE
SET_GCODE_OFFSET Z=0.10 or -0.10 always get that confused.

KevinOConnor, 2018-12-09:
@rmcbc 

1 - The only way to change the z_offset is to edit the config file and run RESTART.  However, as mentioned above, it's generally possible to alter the G-Code z coordinates to similar effect using the SET_GCODE_OFFSET command.

2 - Klipper uses a 2us step/dir pulse time, which works with all common stepper motor drivers.

3 - That's very odd - you'll need to attach a log for us to assist - as described at: https://github.com/KevinOConnor/klipper/blob/master/docs/Contact.md

-Kevin

rmcbc, 2018-12-09:
All my answers clarified. Thanks.

closed 2018-12-09 by rmcbc