Klipper3d/klipper #974: Error during installation on orange pi zero

Ovaday, 2018-12-07:
I am trying to install klipper on my orange pi zero, had already working system with Armbian and OctoPrint.

I did the steps from tutorial:
`cd ~
git clone https://github.com/KevinOConnor/klipper
virtualenv /home/ovaday/klippy-env
./klipper/scripts/install-octopi.sh
cd ~/klipper/
make menuconfig
make`

But during that I got just:
```
ovaday@orangepizero:~/klipper$ sudo make menuconfig
scripts/kconfig/mconf /home/ovaday/klipper/src/Kconfig

*** End of the configuration.
*** Execute 'make' to start the build or try 'make help'.

ovaday@orangepizero:~/klipper$ sudo make
  Build Kconfig config file
#
# configuration written to /home/ovaday/klipper/.config
#
  Creating symbolic link out/board
  Compiling out/src/sched.o
src/sched.c:10:31: fatal error: board/io.h: No such file or directory
 #include "board/io.h" // readb
                               ^
compilation terminated.
Makefile:64: recipe for target 'out/src/sched.o' failed
make: *** [out/src/sched.o] Error 1
ovaday@orangepizero:~/klipper$ 
```

Don't know what's may be wrong.
Don't think klipper log is needed because it's starting without final 'make' and without needed configuration (I had stopped them) during some attempts

[klippy.log](https://github.com/KevinOConnor/klipper/files/2659041/klippy.log)

LucaBertuolo, 2018-12-08:
You have not follow the tutorial...
https://github.com/KevinOConnor/klipper/blob/master/docs/Installation.md

> ovaday@orangepizero:~/klipper$ sudo make menuconfig

Why do you have used "sudo" ?

Ovaday, 2018-12-08:
> Why do you have used "sudo" ?
> And why do you use virtualenv ?
virtualenv just because of another error, I had found it in another issue here. Sudo because of I thought it needed.
> Just follow the tutorial...
Ok, will try reinstall all

Ovaday, 2018-12-08:
Wow, after so many attemps I had removed all the files and services of Klipper, made it directly like in the manual and it works properly!

That's magic!

Solved, thank you :)

Ovaday, 2018-12-08:
Well... 
Now it won't to flash my ramps with arduino mega.
I have did:
```

$ sudo ls -l /dev/serial/by-id/
lrwxrwxrwx 1 root root 13 Dec  8 17:27 usb-Arduino__www.arduino.cc__0042_7543033343735160B1D2-if00 -> ../../ttyACM0
```

```
configured printer.cfg in klipper folder:
[mcu]
serial: /dev/serial/by-id/usb-Arduino__www.arduino.cc__0042_7543033343735160B1D2-if00
pin_map: arduino
```

And the result is:

```
ovaday@orangepizero:~/klipper$ make flash FLASH_DEVICE=/dev/serial/by-id/usb-Arduino__www.arduino.cc__0042_7543033343735160B1D2-if00
  Flashing out/klipper.elf.hex to /dev/serial/by-id/usb-Arduino__www.arduino.cc__0042_7543033343735160B1D2-if00 via avrdude
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: stk500v2_ReceiveMessage(): timeout

```

Ovaday, 2018-12-08:
Also, I had soldered wired connection through tx/rx pins.
I could connect to my marlin firmware by octoprint to /dev/ttyS1 and to /dev/ttyACM0 at baudrate 115200.

In Octoprint:
`Connecting to: /dev/ttyACM0Changing monitoring state from "Offline" to "Opening serial port"Connected to: Serial<id=0xacaf0290, open=True>(port='/dev/ttyACM0', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=10.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitorStarting baud rate detection...Changing monitoring state from "Opening serial port" to "Detecting baudrate"Trying baudrate: 115200Send: N0 M110 N0*125Recv:  T:26.06 /0.00 B:-15.00 /0.00 T0:26.06 /0.00 T1:-15.00 /0.00 @:0 B@:0 @0:0 @1:0`
```

Connecting to: /dev/ttyS1Changing monitoring state from "Offline" to "Opening serial port"Connected to: Serial<id=0xaca4b2f0, open=True>(port='/dev/ttyS1', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=10.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitorStarting baud rate detection...Changing monitoring state from "Opening serial port" to "Detecting baudrate"Trying baudrate: 115200Send: N0 M110 N0*125Recv:  T:27.34 /0.00 B:-15.00 /0.00 T0:27.34 /0.00 T1:-15.00 /0.00 @:0 B@:0 @0:0 @1:0Recv: echo:Unknown command: "\x1b\x01"Recv: okChanging monitoring state from "Detecting baudrate" to "Operational"Recv: okSend: N0 M110 N0*125Send: N1 M115*39Recv: okSend: N2 M21*18Recv: FIRMWARE_NAME:Marlin 1.1.9 (Github) SOURCE_CODE_URL:https://github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:3D Printer EXTRUDER_COUNT:2 UUID:cede2a2f-41a2-4748-9b12-c55c62f367ffRecv: Cap:SERIAL_XON_XOFF:0Recv: Cap:EEPROM:0Recv: Cap:VOLUMETRIC:1Recv: Cap:AUTOREPORT_TEMP:1Recv: Cap:PROGRESS:0Recv: Cap:PRINT_JOB:1Recv: Cap:AUTOLEVEL:0Recv: Cap:Z_PROBE:0Recv: Cap:LEVELING_DATA:0Recv: Cap:BUILD_PERCENT:0Recv: Cap:SOFTWARE_POWER:0Recv: Cap:TOGGLE_LIGHTS:0Recv: Cap:CASE_LIGHT_BRIGHTNESS:0Recv: Cap:EMERGENCY_PARSER:0Recv: Cap:AUTOREPORT_SD_STATUS:0Recv: Cap:THERMAL_PROTECTION:0
!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
Your printer's firmware is known to lack mandatory safety features (e.g.
thermal runaway protection). This is a fire risk.

Learn more at https://faq.octoprint.org/warning-firmware-unsafe
!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!

Recv: okSend: M155 S2Recv: echo:SD init failRecv: okRecv:  T:26.28 /0.00 B:-15.00 /0.00 T0:26.28 /0.00 T1:-15.00 /0.00 @:0 B@:0 @0:0 @1:0Recv: ok
```

Ovaday, 2018-12-08:
Solved.

The tx-rx connection with USB in one time obviously was unnecessary.
After I had unsoldered tx and rx, all worked ok

closed 2018-12-10 by Ovaday

YaronSoffer, 2019-03-20:
You probably needs to do this after the git command 

sudo chown -hR <youruser> *
sudo chown -hR <youruser> .*

in my case:

madorax251, 2021-07-04:
https://github.com/ludiazv/octocitrico/releases/tag/v21.05-1