Klipper3d/klipper #979: How to enable Bltouch

Sapherys, 2018-12-09:
Hello,

First, thank you very much for klipper and all the support, seeing my printer printing at 120 speed ist so satisfying :)

I actually installed a BLTouch Smart physically on my Ender 3, everything is wired. Klipper is allredy installed and updated.

But can you explain me which steps I have to follow to activate it on Klipper and get it work? 

I would be really pleased. Thank you very much!!

BlackStump, 2018-12-09:
https://github.com/KevinOConnor/klipper/blob/master/config/example-extras.cfg#L46

Sapherys, 2018-12-16:
Thank you very much! 

I finally had time to test it and added the mines i thought necessary to enable BlTouch on my Ender 3.

[printer.cfg.txt](https://github.com/KevinOConnor/klipper/files/2683413/printer.cfg.txt)

But after loading this *.cfg file, I have the following error:

```
09:51:22
Connected to host via /tmp/printer @0bps
09:51:22
File contains parsing errors:
[line 109]: 'G4 P200\n'
[line 112]: 'G4 P100\n'
[line 117]: 'G91 ; Uncomment these 2 lines to blindly lift the Z 20mm at start\n'
[line 118]: 'G1 Z20 F600\n'
[line 119]: 'G90\n'
[line 120]: 'G28 X0 Y0\n'
[line 121]: 'G1 X140 Y110 F3600\n'
[line 122]: 'G28 Z0\n'
[line 123]: 'G1 Z10 ; lift to Z10 at the end\n'
Once the underlying issue is corrected, use the "RESTART"
command to reload the config and restart the host software.
Printer is halted
09:51:22
File contains parsing errors:
09:51:22
Firmware version: v0.6.0-668-gef4f84e-dirty
```
I don't understand why, can you please help me to get it right?

Thank you very much!!

KevinOConnor, 2018-12-17:
In order for us to assist, you will need to attach the klipper log file - see: https://github.com/KevinOConnor/klipper/blob/master/docs/Contact.md

Sapherys, 2018-12-17:
Hello Kevin, thank you for your message. Here is the logfile, sorry for not sending it with the last message.

[klippy.zip](https://github.com/KevinOConnor/klipper/files/2686363/klippy.zip)

KevinOConnor, 2018-12-21:
When using a multi-line gcode script in the config file, every subsequent line of the g-code must be indented a couple of space.  So, make sure each of the lines reported in the error above have a few spaces at the start of each line.

-Kevin

KevinOConnor, 2018-12-27:
That error indicates the klipper source code was corrupted.  Try removing the klipper directory and rerunning the steps described in the installation document.

-Kevin

dotps1, 2018-12-31:
@sapherys, did this config work for your ender 3?  I just got a BLTouch sensor and I have no idea where to start to make it work with klipper.

jakep82, 2018-12-31:
@Sapherys The new bltouch module replaces the servo and probe sections.  You also have a discrepancy between pins.  The control pin is the servo pin.  You've defined your servo with PA4, and the bltouch with PA2.  You need to figure out which one is correct, put that in the bltouch section, and then delete the servo and probe sections.

dotps1, 2018-12-31:
@Sapherys I stumbled on this some how:  https://pastebin.com/6e3h2U7N  but it is getting me started, the BLTouch is at least working and stopping the carriage from crashing into the bed.

BlackStump, 2018-12-31:
# vanilla Ender 3 with BLtouch
# This file contains common pin mappings for the 2018 Creality
# Ender 3. To use this config, the firmware should be compiled for the
# AVR atmega1284p.

# Note, a number of Melzi boards are shipped with a bootloader that
# requires the following command to flash the board:
#  avrdude -p atmega1284p -c arduino -b 57600 -P /dev/ttyUSB0 -U out/klipper.elf.hex
# If the above command does not work and "make flash" does not work
# then one may need to flash a bootloader to the board - see the
# Klipper docs/Bootloaders.md file for more information.

# See the example.cfg file for a description of available parameters.

[stepper_x]
step_pin: PD7
dir_pin: !PC5
enable_pin: !PD6
step_distance: .0125
endstop_pin: ^PC2
position_endstop: 0
position_max: 235
homing_speed: 50

[stepper_y]
step_pin: PC6
dir_pin: !PC7
enable_pin: !PD6
step_distance: .0125
endstop_pin: ^PC3
position_endstop: 0
position_max: 235
homing_speed: 50

[stepper_z]
step_pin: PB3
dir_pin: PB2
enable_pin: !PA5
step_distance: .0025
endstop_pin: probe:z_virtual_endstop
position_min: -2 
position_max: 250

[extruder]
max_extrude_only_distance: 100.0
step_pin: PB1
dir_pin: !PB0
enable_pin: !PD6
step_distance: 0.010526
nozzle_diameter: 0.400
filament_diameter: 1.750
heater_pin: PD5
sensor_type: EPCOS 100K B57560G104F
sensor_pin: PA7
control: pid
# tuned for stock hardware with 200 degree Celsius target
pid_Kp: 21.527
pid_Ki: 1.063
pid_Kd: 108.982
min_temp: 0
max_temp: 250

[heater_bed]
heater_pin: PD4
sensor_type: EPCOS 100K B57560G104F
sensor_pin: PA6
control: pid
# tuned for stock hardware with 50 degree Celsius target
pid_Kp: 54.027
pid_Ki: 0.770
pid_Kd: 948.182
min_temp: 0
max_temp: 130

[fan]
pin: PB4

[mcu]
serial: /dev/ttyUSB0

[printer]
kinematics: cartesian
max_velocity: 300
max_accel: 1000
max_z_velocity: 5
max_z_accel: 100

[bltouch]
sensor_pin: ^PC4
control_pin: PA4
pin_move_time: 0.200
x_offset: -43
y_offset: 4
z_offset: 1.58
speed: 5

[homing_override]
axes: z
set_position_z: 5
gcode:
    G90
    G1 Z10 F600
    G28 X0 Y0
    G1 X160 Y120 F3200
    G28 Z0
    G1 X0 Y0 F7200

[display]
lcd_type: st7920
cs_pin: PA3
sclk_pin: PA1
sid_pin: PC1
encoder_pins: ^PD2, ^PD3
click_pin: ^!PC0

dotps1, 2018-12-31:
so do I need to call a G28 in my start up GCode?  or should that homing override be called during every print start up?

KevinOConnor, 2019-01-10:
I'm going to close this as I think the original question was answered.

-Kevin

closed 2019-01-10 by KevinOConnor