Klipper3d/klipper #980: Strange problem with power off / power on

rmcbc, 2018-12-09:
My system:
- Arduino mega + RAMPS 1.4 (USB don't power up arduino, the cable don't have +5V, I remove it)
- Raspberry pi 2
- Power sypply to arduino + ramps controlled by relay using GPIO in raspberry
Problem: 
 System is all ok, arduino mega powered and connected to Octoprint.
 In Octoprint shutdown arduino + ramps power
 In Octoprint power up arduino + ramps
 Serial connect to klipper
 Printer says not ready and display doesn't show information.
 In terminal do FIRMWARE_RESET and no printer ready.
 Change in [mcu] restart_method: to arduino, command and rpi_usb none solves the problem.
Solution:
 Take the USB cable from Raspberry to Arduino and put it again, then the printer became ready.

Can some one explain to me what I'm doing wrong?

BlackStump, 2018-12-10:
without a log file there is not much info to go on.
but as a guess your pi is not finding the usb, do you have usb by id if not that might be the solution
https://github.com/KevinOConnor/klipper/blob/master/docs/FAQ.md#wheres-my-serial-port

https://github.com/KevinOConnor/klipper/blob/master/docs/Contact.md

rmcbc, 2018-12-12:
Solve using "/dev/serial/by-id/...."
Thanks.

closed 2018-12-12 by rmcbc