Klipper3d/klipper #981: Homing suddenly doesn't work anymore (Safe Z homing)

radry, 2018-12-09:
I sucessfully used klipper before with the same config, then didn't use my printer for several months.
Now homing doesn't work anymore when I click "home" in octoprint webinterface.

It homes X and Y but not Z with following error:

```
Send: G91
Recv: ok
Send: G28 X0 Y0
Send: G90
Recv: !! Must home axis first: 100.000 130.000 -0.362 [0.000]
Recv: !! Must home axis first: 100.000 130.000 -0.362 [0.000]
```

Homing script in config:

```
[homing_override]
gcode:

    G28 X0 Y0            ; home X and Y
    G0  X100 Y130       ; move to safe homing spot
    G28 Z0               ; home Z
    G0  Z10              ; clear homing button
```

What is wrong or how do I do a "safe homing" of Z?

BlackStump, 2018-12-10:
suggestion is to try G28 in the Octoprint terminal, and as always include the log.

https://github.com/KevinOConnor/klipper/blob/master/docs/Contact.md

radry, 2018-12-10:
Here you go, exact same result.

G28 from terminal and 

```
!! Must home axis first: 100.000 130.000 -0.362 [0.000]
!! Must home axis first: 100.000 130.000 -0.362 [0.000]
```

[klippy.log](https://github.com/KevinOConnor/klipper/files/2664710/klippy.log)

jakep82, 2018-12-10:
Try adding the following to your stepper_z1 section:

> position_min: -1

jakep82, 2018-12-10:
Actually, looking closer it seems you're missing something in your homing override section.  You need the line below.

> set_position_z: 5

KevinOConnor, 2018-12-21:
It looks like this question was answered, so I'm closing this issue.

closed 2018-12-21 by KevinOConnor