Klipper3d/klipper #986: [Feature request] Seperate accle and Velocity for Extruders

Ahrenfelde, 2018-12-10:
Hi,

it would be nice to seperate the acceleration and velocity from the main axis an the extruder.

I have a bowden setup, there is a fast retract nessesary for a good print quality. But actual it is not possible to crank the accle up, because my printhead is to heavy for value above 4000.

Greetings

jakep82, 2018-12-10:
https://github.com/KevinOConnor/klipper/blob/master/config/example.cfg#L118

Ahrenfelde, 2018-12-10:
Yes, i now this Parameter, but it is only a maximum or not?
I want to bring my extruder to a higher Accel than my axis. 
As example 
max_accle : 2000 
max_e_accle: 7000

With the max_extrude_only_accel parameter is is only possible to reduce the accleration from 2000 to zero or do i missunderstood #892

motorist828, 2018-12-12:
Many people have problems due to the fact that the extruder acceleration depends on the overall value
This problem manifests itself even on direct extruder
I conducted the test, high acceleration completely removes the seam, but the printer is noisy at the same time
If the acceleration is normal, then the extruder is too slow and the seam is ugly.
[image]

Make printer and extruder accelerations independent really need

St007, 2018-12-12:
+1 seems like we need acceleration limits to be more clear for end user : if one uses max_e_accel it must be only limit applied for extruder and no other acceleration limits shouldn't  be applied

Ahrenfelde, 2018-12-12:
Hi,

yes i tested it also with higher accle for the whole printer. But my printer gets not really noisier. 
I get really bad ringing artefacts with high accle rates. The picture shows a mold core for a silikon sock for a E3D V6. (added the picture later on)

This is a problem with klipper and i wondering why here are not more issues about this topic. 
All the "standalone" Firmwares like Marlin or Repetier gets the option to tune the Extruder seperatly.

jakep82, 2018-12-12:
> This is a problem with klipper and i wondering why here are not more issues about this topic.

I think for most people it works fine as-is.  It does for me.

Ahrenfelde, 2018-12-13:
Please don't understand me wrong. Klipper is very fine, it works great but here and there are some points there are not 100% fine, exept for me. I opened this request to discuss this topic not to bash klipper.

KevinOConnor, 2018-12-13:
This looks like a duplicate of #892.  If someone wishes to implement this, then that's fine.

-Kevin

Ahrenfelde, 2018-12-14:
okay, maybe here are a good programmer who are able to imlement this. My programming skills with python are where low so i can´t do this at the moment.

Ahrenfelde, 2018-12-24:
okay i'm now a little bit confused.

At the moment i am trying to tune pressure advanced. I had seen that the "retract" from pressure advanced will move very faster than the normal retract from slicer. The Settings in my Slicer are 50mm/s and 4mm retracten lenght. 
Now i define max_extrude_only_velocity and max_extrude_only_accel to my normal maximum settings for the whole pinter in my cfg. Now the retract is quite faster than befor. It was the whole time blocking from the standart settings of klipper when these two settings are not definded. But why?

Now i will first tune my retract setting again to see of a faster movement will bring a improvment in print quality.

muenznej, 2018-12-25:
hi,
I dont know if I understand your question correctly but the speed and accelerations of the extruder are calculated per default like this:

```
def_max_extrude_ratio = def_max_cross_section / self.filament_area
        max_cross_section = config.getfloat(
            'max_extrude_cross_section', def_max_cross_section, above=0.)
        self.max_extrude_ratio = max_cross_section / self.filament_area
        logging.info("Extruder max_extrude_ratio=%.6f", self.max_extrude_ratio)
        toolhead = self.printer.lookup_object('toolhead')
        max_velocity, max_accel = toolhead.get_max_velocity()
        self.max_e_velocity = config.getfloat(
            'max_extrude_only_velocity', max_velocity * def_max_extrude_ratio
            , above=0.)
        self.max_e_accel = config.getfloat(
            'max_extrude_only_accel', max_accel * def_max_extrude_ratio
            , above=0.)
```
but i dont think(know?) that PA and "normal" retractions do things differently.

For me that ratio is 0.2661 so i have defaults of 
max_extrude_only_accel ~ 800 
max_extrude_only_velocity ~ 80
with my 3000 mm/s2 and 300mm/s global printer limits.

If u set max_extrude_only_accel  and max_extrude_only_velocity manually you can set them as higher than 80/800 but only up to 300/3000 ( in my case) (ref: https://github.com/KevinOConnor/klipper/issues/892#issuecomment-439436288)

Hope that helps ( and that it is correct :D )

I also upvote the separation of these limits for XY vs E but i cannot contribute due to a lack of skill :sob:

KevinOConnor, 2019-01-04:
FYI, the main printer accel/velocity should no longer be applied to extrude only moves (commit 24fe606d).

-Kevin

muenznej, 2019-01-07:
thx you are awesome!

Ahrenfelde, 2019-01-08:
thx Kevin! You are great!!!

closed 2019-01-16 by KevinOConnor

Spacemarine2018, 2020-02-21:
I am confused because I'm getting inconsistent behaviour from Klipper v0.8.0

If I specifiy max_extrude_only_accel, will any m204 Sxxxxx command in front of an extrude-only move be ignored? Or is max_extrude_only_accel only the upper limit that can be set via m204 ?

And the same goes for max_extrude_only_velocity. Which value will be used if I specify it in the config and also via G1 Exy fXY?