Klipper3d/klipper #987: Manual bed mesh leveling improvement

rlesac, 2018-12-10:
As my ABL is not working and I'm still waiting on replacement, I have to use the manual mesh bed leveling.

It would be very helpful to have support for continuous z movement in discrete steps.

Currently there is no interface support for it, so I edited the menu.cfg:
`[menu __bedmesh __movez0025mm]
type: input
enable: !toolhead.is_printing
name: "MZ 0.025:{0:05.3f}"
parameter: toolhead.zpos
input_min: -1.0
input_max: 5.0
input_step: 0.025
input_step2: 0.100
gcode: G1 Z{0:.3f}
`
I also added menu items for start, next and save.

That did not work, as moving the z axis in uneven steps, gave garbage results.
The only way I could get a useful mesh is by moving in discrete steps, ie. press button, move 0.025, press again, repeat ad nauseam. 

Same with corner leveling.
Very tedious.

Also, if the scroll slipped and instead of moving z went to the next point, I had to start all over again.

In Marlin, it just a simple routine, scroll to move z, press to go to the next point.
I'm sure it would be very useful to a lot of people.

reefwarrior, 2018-12-10:
Can you perhaps also give the menu code for 'start, next and save'? Also having issues getting ABL to work, so this will save me a ton of time!

reefwarrior, 2018-12-10:
Understand now what you mean with this being a bit tedious... need more of a 'live' method as with marlin.

KevinOConnor, 2018-12-17:
FYI, this is being worked on - see #741.

-Kevin

KevinOConnor, 2019-01-06:
I'm going to close this in favour of tracking at PR #741.

-Kevin

closed 2019-01-06 by KevinOConnor

ardyesp, 2019-07-21:
I know a better mesh bed leveling is being worked on. Meanwhile, here is my configuration entries, in case someone else wants to use a graphic display to manually generate the mesh.

```
[stepper_z]
position_min: -0.5

[bed_mesh]
min_point: 20,20
max_point: 230,230
probe_count: 3,3
fade_start: 1.0
fade_end: 10.0

	
[menu __prepare meshBed]
type: list
enable: !toolhead.is_printing
name: Mesh Bed Level
items:
    .start
    .zposition
    .zNudgeDown
    .zNudgeUp
    .next
    .save

[menu __prepare meshBed zposition]
type: item
name: "Z Pos= {0:05.3f}"
parameter: toolhead.zpos
	
[menu __prepare meshBed start]
type: command
name: Start Callibration
gcode: 
	G28
	BED_MESH_CALIBRATE
	TESTZ Z=-5

[menu __prepare meshBed zNudgeDown]
type: command
name: Nudge Down
gcode: TESTZ Z=-0.025

[menu __prepare meshBed zNudgeUp]
type: command
name: Nudge Up
gcode: TESTZ Z=0.025

[menu __prepare meshBed next]
type: command
name: Next
gcode: 
	ACCEPT
	TESTZ Z=-5
	
[menu __prepare meshBed save]
type: command
name: Save & Restart
gcode: SAVE_CONFIG
```

s1mpleman, 2019-08-03:
> 
> 
> I know a better mesh bed leveling is being worked on. Meanwhile, here is my configuration entries, in case someone else wants to use a graphic display to manually generate the mesh.

Your a legend thanks for this! My bed is warped to hell now and was just looking up best way to generate a mesh.

mickw99, 2019-09-15:
followed this step for step.  the issue i'm having is when i nudge up or down.  it goes up then right back to where it was.

ardyesp, 2019-09-15:
Net move of a nudge command moves the bed by 0.025mm. You should see the new Z value in the menu item right above nudge, like Z Pos= -0.05 etc.

mickw99, 2019-09-15:
says it moves but its still in the same position.  like i said it moves up then right back to the same spot.  doesnt matter if if nudge up or down.

The-Holgi, 2020-05-14:
Hello,
works this after new start of the printer, or must leveling after every new start?
Where the „meshdata“ after leveling stored?

ardyesp, 2020-05-14:
If you choose "SAVE_CONFIG" the mesh data is written back to your configuration file (at the end, as a comment block) and printer restarts. It will automatically be loaded upon next restarts.

The-Holgi, 2020-05-14:
Hm, after leveling and chose save in the lcd menu, there is no mesh data in my printer.cfg.

RicardoCB76, 2020-09-22:
Hello, a great tutorial, but I have a question, after saving the leveling, how do you load it if you want to print something, do you have to execute a command? or does nothing need to be done?

Is that when she had Marlin she had to put the command M421 S1 after G28.

And now I have the doubt of how it is done.

Thank you

ardyesp, 2020-09-22:
> 
> 
> If you choose "SAVE_CONFIG" the mesh data is written back to your configuration file (at the end, as a comment block) and printer restarts. **It will automatically be loaded upon next restarts**.

RicardoCB76, 2020-09-22:
Thanks for the quick response, another thing I wanted to comment on, when you prepare the piece in the laminator (Simplify3D) the initial or final script which would be the most suitable, or it is not necessary to put anything, I comment on the subject of the M420 code S1 it needs to be included in Klipper.

ardyesp, 2020-09-22:
Its not necessary to add anything in slicer start gCode.

RicardoCB76, 2020-09-22:
Ok, thanks again, there was no way of knowing, I have been looking on the web and could not find the solutionEqw

thelastWallE, 2020-09-27:
```
[menu __prepare meshBed zposition]
type: item
name: "Z Pos= {0:05.3f}"
parameter: toolhead.zpos
```
This gives me now errors in the klipper.log:
`error: Choice 'item' for option 'type' is not a valid choice`

ld3300, 2020-09-27:
try changing __meshBed __zposition

er9xuser, 2021-01-22:
Hi. I know it is old post..but i did everything per "thelastWallE" last post...but where to find this menu now? On LCD...i got nothing.

Thanks

coolo, 2021-02-27:
Replace `__prepare` with `__setup` and you will find it as last menu item below Setup. `__prepare` is no more in current klipper