Klipper3d/klipper #989: Question about STM32F103

anwofis, 2018-12-11:
Hello,

i'm just building a new 3d printer and i am planning to make the electronics modular (i'll use external drivers and some Arduino modules for sensors and MOSFETs) and ofc use Klipper as firmware.

As MCU i want to use a STM32F103RBT6 as it has some more pins than the C-version.

Is this MCU supported? 

(If yes: Do the V- and Z-versions also work?)

Best regards,

Andreas

pamalofeev, 2018-12-11:
Directly about such support is not stated, but ... For stm32f103cbt6, stm32f103rbt6 and stm32f103vbt6 the same datasheet. For them, the position of alternative functions for pins coincides. It seems that this is the same chip, but in different cases. The makefile says "CFLAGS + = -DSTM32F103xB", this also indirectly shows the identity of these chips. I would try if I were you. ;-)

anwofis, 2018-12-11:
This is good news, thank you. I already ordered a STM32F103RBT6 and will try Klipper on it, when it arrives.

anwofis, 2018-12-12:
Too bad. I'll use an Arduino MEGA then until this is fixed. (I'm planning to use two PT100 + MAX31865 and some TMC2130 with SPI.) I have got one working already with Klipper ;)

KevinOConnor, 2018-12-27:
It looks like the original question was answered, so I'm going to close this issue.

-Kevin

closed 2018-12-27 by KevinOConnor

anwofis, 2019-01-29:
I didn't want to open a new issue, so i reopen this one just for future reference:

I installed klipper on a STM32F103RBT6 and it's working fine (SPI/I2C is untested.)

All pins seem to work!

On Ubuntu 18 i coudn't compile klipper first. I had to install the following packages manually:

sudo gdebi libnewlib-dev_3.0.0.20180802-2_all.deb 
sudo gdebi libnewlib-arm-none-eabi_3.0.0.20180802-2_all.deb 

Ubuntu 18 ships with version 2.4.x.

I then flashed it via the [stm32duino bootloader](https://github.com/rogerclarkmelbourne/STM32duino-bootloader/tree/master/binaries) (the generic one - PC13). I just pressed the reset button before flashing.