Klipper3d/klipper #990: Can't make OctotPrint printer ready

LeonKnook, 2018-12-11:
Did the hole procedure and are almost there, but don't get printer ready in octoprint.
Commented out al the heaters to avoid there is a thermistor on the wrong pin.

Send: status
Recv: // Klipper state: Ready 

[klippy.log](https://github.com/KevinOConnor/klipper/files/2669353/klippy.log)

Please some help

kind regards,
Leon

jakep82, 2018-12-11:
Not sure what issue you're seeing.  When sending status the printer is returning ready.  That would indicate it's connected and ready to print.  The log looks normal.

LeonKnook, 2018-12-11:
Good to know but should there be the message ' printer ready'  en not ' klipper state ready' ?
did the whole procedure again, flashing the firmware and all that.

Send: firmware_restart
Recv: ok
Recv: // Klipper state: Disconnect
Recv: // Klipper state: Ready

Thank you for your effort.

jakep82, 2018-12-11:
That message is exactly what you want to see. Klipper runs the printer.  The printer is ready.

LeonKnook, 2018-12-11:
Must be an issue with the radds board maybe, arduino due is brand new. Home all does not work, no movement. Endstops are in correct logic.

jakep82, 2018-12-11:
Have you tried stepper_buzz to see if the motors work?  Do your heaters work?

https://github.com/KevinOConnor/klipper/blob/master/docs/Config_checks.md

LeonKnook, 2018-12-12:
Thank very much. I will do a complete walk true on that document and come back.

Hywelmartin, 2018-12-12:
is it a clone Due?   #844

LeonKnook, 2018-12-12:
Yes it is a clone Due. And I flashed wirt ttyACM0 which is there in the config.cfg

LeonKnook, 2018-12-12:
That document was perfect! It solved all my problems. Not yet printed but everything is working properly now.
Thank very much for pointing me in the right direction.
Kind regards,
Leon

KevinOConnor, 2018-12-21:
It sounds like this issue was resolved, so I'm closing it.

closed 2018-12-21 by KevinOConnor