Klipper3d/klipper #991: Unable to Flash Prusia i3 MK3

fhgui, 2018-12-12:
I've been struggling for about a month now trying to flash the micro-controller on my Prusa I3 MK3. The only way I've ever been successful with flashing the micro-controller was by using Slic3r PE. If anyone knows a way to flash the Klipper using Slic3r PE that would be greatly appreciated. I've already tried transferring the .hex file using filezilla then flashing using Slic3r PE. The flash was successful but didn't work. (I assumed this would happen) I then had to flash the stock firmware. Here are the ssh commands. If there are only other logs that I can attach to help please let me know. 
pi@octopi:~ $ cd ~/klipper/
pi@octopi:~/klipper $ make menuconfig
scripts/kconfig/mconf /home/pi/klipper/src/Kconfig

*** End of the configuration.
*** Execute 'make' to start the build or try 'make help'.

pi@octopi:~/klipper $ make
make: Nothing to be done for 'all'. // I previously compiled but did it again to show I did every step
pi@octopi:~/klipper $ sudo service klipper stop
pi@octopi:~/klipper $ sudo make flash FLASH_DEVICE=/dev/ttyAMA0 // I have already verified /dev/ttyAMA0 is correct 
Flashing out/klipper.elf.hex to /dev/ttyAMA0 via avrdude
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: stk500v2_getsync(): timeout communicating with programmer

avrdude done.  Thank you.

src/avr/Makefile:32: recipe for target 'flash' failed
make: *** [flash] Error 1

pi@octopi:~/klipper $

yschroeder, 2018-12-12:
The MK2 has a bootloader and the MK3 should have one as well. How else would you be able to upgrade the printer firmware directly via USB?

Look for the correct avrdude command (should be the same that is used to upgrade the stock firmware).

jakep82, 2018-12-12:
https://mobile.twitter.com/josefprusa/status/940725283900084225?lang=en

yschroeder, 2018-12-12:
They got rid of the Arduino bootloader, they use something different instead.

Try this command line:
```
avrdude -v -p m2560 -c wiring -P /dev/ttyACM0 -U flash:w:out/klipper.elf.hex:i -D
```

NordicGuy, 2018-12-12:
Which pi have you installed Klipper on? 
And are you trying to flash via USB or the UART pins directly into the Einsy board? 
A number of people have successfully flashed their MK3’s over on the PrusaOwners discord channel: https://discord.gg/d3kzPqN

fhgui, 2018-12-12:
@yschroeder Tried the listed command and got the same results.
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: stk500v2_getsync(): timeout communicating with programmer

@jakep82 I don't have a device to flash a bootloader itself but I will look into doing this with the PI. If that doesn't work I won't be able to purchase a avr isp for a month or so due to Christmas and other expenses but that is something i'll have to try down the road. 

@NordicGuy I'm using a PI 0 /W connected directly with UART pins. I plan on getting a PI 3 B+ if I run into CPU issues running klipper on the PI 0 but that's something I plan on purchasing down the road. I will try asking the discord for help over the weekend when I have more time to test things. 

Thank you everyone for the quick responses and helpful advice!

NordicGuy, 2018-12-13:
> @NordicGuy I'm using a PI 0 /W connected directly with UART pins. I plan on getting a PI 3 B+ if I run into CPU issues running klipper on the PI 0 but that's something I plan on purchasing down the road. 

There’s your problem. The Prusa bootloader doesn't support flashing the firmware via the UART pins. A custom bootloader has been discussed on discord, but not much has come of it. You need to first flash the Einsy via USB and then set up UART communication. The guys over in discord are sure to want to help in that regard.
Beware that the Pi0W has very little computational overhead running octoprint, so you’ll likely need to set up the Virtual-SD-card feature to get Klipper running smoothly.

KevinOConnor, 2018-12-21:
It sounds like the original question was answered, so I'm going to close this issue.

closed 2018-12-21 by KevinOConnor