Klipper3d/klipper #993: BLTOUCH?!

xsasx, 2018-12-12:
Sorry im new to klipper so please apologize my question.

Klipper running smooth after a hard learning phase (come from marlin)

I have an Anet A8 and ordered an Bltouch from Trianglelab.

Does exist any how to how to install it and configure it to use it with Klipper.

I have read here very much but its more or less deep dive information from experts :/

Maybe someone can help me out how to start step by step.

1. I will print one of this (https://www.thingiverse.com/thing:2773637  // https://www.thingiverse.com/thing:1846913 // https://www.thingiverse.com/thing:2091529)

2. wire it up (think i can handle also this part)

3. and than ? !

jakep82, 2018-12-12:
https://github.com/KevinOConnor/klipper/issues/979

https://github.com/KevinOConnor/klipper/issues/955

xsasx, 2018-12-13:
Thanks ! I have read all of this - helped not that much - would be good to have aguid like the pressure advanced guide 👍

KevinOConnor, 2018-12-27:
There's some bltouch documentation in the config/example-extras.cfg file.  It doesn't look like anyone here has detailed information on configuring the anet a8, so I'm going to close this issue.

-Kevin

closed 2018-12-27 by KevinOConnor