Klipper3d/klipper #994: Underextrusion after small skin using Pressure Advance

arbitur, 2018-12-13:
Ive been trying to get Klipper to work perfectly for a couple of weeks now and it seems to be working OK, however Im still having some problems with Pressure Advance when trying to print the [xyzCalibration_cube](https://www.thingiverse.com/thing:1278865). I keep getting underextrusion at the start of new layers after the lower and upper skins (4 layers thick) are printed for the X and Y letters:
[image]

I have tried `pressure_advance: 0...1` and my edges looks good at ~1.9 so thats the setting Ive settled for. Ive tried with many different `pressure_advance_lookahead_time: 0...0.5` but nothing seems to work.

When I look at the extruder while printing these small skins I can see that it extrudes and retracts to the same spot meaning the net move is pretty much 0 and thus not enough filament is pushed out which reduces pressure by a lot and making it not push out anything for 1-3mm and then it takes ~20mm to gain the correct pressure for a whole line width.

[image]

Any help/tips?

arbitur, 2018-12-13:
> 1.9 is an incredibly high number. I would venture a guess that your extruder is either missing steps or grinding filament at a value that high.

Oh shit Im sorry I meant to say 0.19... Ill edit my post.

BlackStump, 2018-12-13:
because there is no log file, and no info about your printer and how it is configured all I can give is very generic suggestions.
It looks to me to be a extruder issue ie as @jakep82 has mentioned skipping steps or grinding filament.
why it is doing that needs to be explored.
Try printing the cube at 20mm/s or less, try also increasing temp.
look at the extruder mechanics ie is the enough spring pressure pushing the filament against the drive wheel are you just trying to print too fast for that style of extruder, some of the out of the box printers work fine provided you do not print fast.
Fine tuning extruders with pressure advance will only work if the extruder is in the first place functioning correctly.
Very easy to get caught up in print quality issues only to find out that the extruder has been the cause of it.
one way you can tell if there is a extruder issue is to mark 50mm of filament then extrude 50mm through the hotend slowly and see that infact you extruded 50mm then do the same test extruding faster.

arbitur, 2018-12-13:
@BlackStump Sorry, I wanted to keep it short since it would get long.

My printer is a modified Tevo Tarantula with V6 full metal hotend, Titan extruder (1:3 gear ratio) and Nema17 motor for extruder, capricorn bowden tube ~35cm long,, extruder steps is recently calibrated.
Printing in PLA at 215C, Cura slicer, 100mm/s printing speed, 50mm/s for outer walls and skin.

Where can I get these logs you are mentioning?

BlackStump, 2018-12-13:
you will find the log with the help of this doc.
https://github.com/KevinOConnor/klipper/blob/master/docs/Contact.md

I would still try printing slower if only to see if you get the same results. Your printer is certainly capable of printing at higher speeds but you need to start somewhere to help diagnose the problem.
looking at the ringing on your print you may want to lower your acceleration.

a quick search seem that the titan extruder doesn't like high extruder acceleration or high retraction speed ....that might be also a place to start

KevinOConnor, 2018-12-13:
I would not recommend tuning pressure advance with an xyz calibration cube.  I recommend using the simple square stl and following the guide at: https://github.com/KevinOConnor/klipper/blob/master/docs/Pressure_Advance.md .  Specifically, the xyz cube results are easily confused with other printing issues (link ringing), and the xyz cube may not have sufficiently long "run ways" to fully build up pressure in the extruder.  If you still have issues tuning pressure advance after following the above guide, then please post some pictures of the square and be sure to attach a klipper log file as described at: https://github.com/KevinOConnor/klipper/blob/master/docs/Contact.md

-Kevin

arbitur, 2018-12-15:
@KevinOConnor I've read the Pressure_Adcance document carefully before I printed my xyzCube and I came to 0.19. I also did the Marlin k-value tool and just replaced the M900 to SET_PRESSURE_ADVANCE ADVANCE=0.0...1.0 and came to the same conclusion of 0.19. So Im pretty sure my pressure advance value is correct. Its just that it seems to be the extruder motor extruding/retracting the exact amount for all those short lines so the material just oozes out and the nozzle gets empty for the next wall. 

@BlackStump I did try to print at 50 mm/s (25 outer and skin) and got the same result as with 100.

arbitur, 2018-12-15:
@KevinOConnor The document doesnt say How to print the test print (square.stl). How many walls (perimeters)? How much infill? What type of infill? Skin?

arbitur, 2018-12-16:
[klippy.log](https://github.com/KevinOConnor/klipper/files/2683825/klippy.log)
[image]
[image]

This log file is for the object above, you can see the gaps at the start of the line.

arbitur, 2018-12-16:
@KevinOConnor I did try a retraction distance of 0.5mm before and just tried with 1mm now but its the same result. The retractionspeed I use (in my slicer) is 40mm/s. But Ill try without retractions. What is your recommendation for Combing (Traveling across already printed lines instead of retraction)? I know in Linear Advance (Marlin) they recommend against Combing, how would this work with Klipper?

arbitur, 2018-12-16:
Oh god I feel so stupid... Cura fooled me, I have retractions settings in my material profile but you can still change global retraction settings which overrides the profile and somehow without me editing the global setting at all it didnt update so my retraction distance was set to 3.5mm -.- Now When I finally realized this bug in Cura and made it so the Actual retraction distance is 0.5mm it works great now...

Im sorry for wasting all your time. You can close this thread.

KevinOConnor, 2018-12-21:
It looks like this issue was resolved, so I'm closing this issue.

closed 2018-12-21 by KevinOConnor