Klipper3d/klipper #996: Want to adjust homing

LeonKnook, 2018-12-13:
Busy with configuration. Homing buttons work fine but when I push a home button I want the Z going 15mm up first before it travels. Reason: To avoid bounching against clips.  How can I adjust that somewhere in the config.cfg?

jakep82, 2018-12-13:
I would suggest reading the entire example-extras.cfg file.  It contains a ton of information.  This can be accomplished with a homing override.

https://github.com/KevinOConnor/klipper/blob/master/config/example-extras.cfg#L844

jakep82, 2018-12-14:
[homing_override]`
set_position_z: 5
gcode:
G0 Z7
G28 X0 Y0
G0 X100 Y100
G28 Z0

This will move Z up 2mm, home X and Y, move X and Y to 100,100, then home Z.  This is what I use to home my printer with a bltouch for Z in the middle of the bed.  Adjust as necessary

LeonKnook, 2018-12-14:
Then in my situation it should be

[homing_override]
set_position_z:20
gcode:
 G28 X0
 G28 Y0
 G28 Z0

But it does not work

jakep82, 2018-12-14:
You aren't understanding how this works.  Just please copy my code below.  I would personally recommend against blindly lifting Z by 20mm because you risk damaging your printer in a situation where you home the printer when it's near the top of it's Z travel.

[homing_override]
set_position_z:5
gcode:
G0 Z25
G28 X0
G28 Y0
G28 Z0

LeonKnook, 2018-12-15:
Thank you that works!
I know the risk of hitting the max z-height. But in my case with my Tronxy X3S the max height is 40 cm/15.75 inch. I never print models near that height. Still don't understand the logic in the code.
set_position_z:5 suggest for me there will be a first move for the z axes to go up 5mm. But never mind it works. If got everything working now except for a method of bed leveling without a probe and have e look at the possibility to equal differences with two z-motors. Thanks again.
Leon.

Arksine, 2018-12-16:
set_position_(xyz) forces the current position to whichever axis you specify.  When I use it I typically set it to 0, I see no real reason to do otherwise.  For example, if you want to move the tool up 20mm on the Z axis as i your configuration, you would do the following:

[homing_override]
set_position_z: 0
gcode:
G0 Z20
G28 X0
G28 Y0
G28 Z0

The G0 Z20 command will move the tool 20mm up because the absolute distance from the current z position (0) is 20.