Klipper3d/klipper #997: Documentation of virtual end stop of TMC2130

yschroeder, 2018-12-14:
I think the documentation in `example-extra.cfg` on the TMC2130 virtual end stop feature is insufficient. I needed the help of the Trinamics datasheet to figure it out.

First of all, there should be some info that the `DIAG1` pin of the driver is configured as open collector. This means, that some pull up resistor is needed on the line. There should be a hint to enable the internal pull up of the MCU if none is present on the PCB. Then, there should also be the info that the pin is active low. So you basically end up with `diag1_pin: ^!PIN`.

Next, it would be helpful to have some info for tuning the `driver_SGT` value. Like:

Start with a value of 0. Try to home the axis. If the motor does not stop when it hits the mechanical limit of the axis, decrease the value of `driver_SGT` to increase the sensitivity (make it trigger earlier). Otherwise, if the motor stops before hitting the mechanical limit and you get the error message `Endstop x still triggered after retract`, then the SGT value is too low. Increase it a bit. The minimum allowed value is -64, the maximum is 63. You have arrived at the optimal value if the axis bumps very gently into the mechanical limit and homing works reliably.

I am willing to improve the documentation, I am not sure though where the best place is. Either directly in `example-extra.cfg` or in a new file in the `docs` folder. Then, there should be a reference to the extra documentation in `example-extra.cfg`.

I only have the TMC2130 available, but I guess that the other Trinamics drivers work similarly. So maybe the documentation should cover all of them at once?

KevinOConnor, 2018-12-14:
I agree that a new document in docs/ sounds like the best strategy.

-Kevin

yschroeder, 2018-12-16:
I looked at the three Trinamics drivers Klipper has implented: 2130, 2208, and 2660. The 2208 does not have the stallguard feature so it cannot be used for sensorless homing. The 2660 has stallguard, but its use for homing is not implemented in Klipper yet.

The main difference of the 2660 is that it has a dedicated output for stallguard which is push/pull logic and cannot be configured to anything else (neither open-collector).

The only board that uses the 2660 is the Duet board, otherwise it does not seem to get too much love from the 3D printing community. I guess mostly because the TQFP chip does not fit onto a stepstick sized PCB...
So I'm not sure how important it is to implement sensorless homing for the 2660.

The 2130 on the other hand has some more options that might be useful to Klipper users: The stallguard output can be configured to be on DIAG0 and/or DIAG1 and both pins can be configured to be push/pull logic, so Klipper could offer options to configure those features.

I think I will write a guide to configure sensorless homing for the 2130 and it can be extended later for the 2660, once it is supported.

KevinOConnor, 2018-12-27:
Lets move any further discussion to PR #1004.

-Kevin

closed 2018-12-27 by KevinOConnor