Klipper3d/klipper #1003: e_step_distance help

dotps1, 2018-12-16:
Im trying to calibrate my e_step_distance, using this link from the help docs: https://reprap.org/wiki/Triffid_Hunter%27s_Calibration_Guide#E_steps

so, the default ender 3 step distance is `0.010526`, after putting my printer in relative mode `M83`, and extrude out 100mm: `G1 E100 F100`, that leaves me with 35mm to my 120mm mark, so using the formula from the link `0.01526 * (100 / (120 - 35))` and round that to the sixth decimal, i get `0.012348`.  if i then change my e_step_distance to that, and run the test again, it gets worse, its no 47mm to my 120mm mark.  if i run that formula `0.12348 * (100 / (120 -47))`, that gives me `0.016964`.  if i use that value and run the test a third time, it gets worse yet.

any ideas what i am doing wrong here?

yschroeder, 2018-12-16:
Klipper configuration is in mm/step, while other firmwares use steps/mm.

So `0.010526 * ((120 - 35) / 100) = 0.0089471` should be the value you are looking for.

dotps1, 2018-12-16:
ah, good to know!  thanks

dotps1, 2019-01-22:
So, I was printing, and looked like it was under extruding, so I test it, and there was 40mm left to my mark of 120mm after extruding 100mm.  so I used that formula and got a smaller `step_distance` value, plugged that in, got worse.  I don't get this.  I tried replacing the nozzle, I re set my extruder gear.  should I not be calibrating with the nozzle on?

jakep82, 2019-01-22:
I have a bowden and calibrate with the bowden tube disconnected.  I simply push filament through the tube until it's flush at the end of the tube, extrude 100mm, and measure how much is now sticking out beyond the end of the bowden tube.

For your specific setup, it's not really possible to help without a log file to see what is happening.

https://github.com/KevinOConnor/klipper/blob/master/docs/Contact.md#issue-reporting

dotps1, 2019-01-22:
i have a Bowden as well, perhaps ill give that a try after this print.  i set it back to the default value and its printing better.

KevinOConnor, 2019-02-03:
I'm going to close this as it looks like the issue was resolved.

-Kevin

closed 2019-02-03 by KevinOConnor

dotps1, 2019-03-17:
I know this is closed, but I just wanted to add this, maybe add to troubleshooting steps doc???  Anyway, upon pulling my Bowden tube to do what @jakep82 suggested, I found the fitting was pitching my Bowden tube like crazy.  I calibrated and found that the default value was perfect at 100mm.  so I cut that pitch out, put a new fitting on, and its printing amazingly, with the default value from the ender-3 config file.