Klipper3d/klipper #1005: Installing Klipper on TinkerBoard

welo78, 2018-12-17:
Hello I am trying to install Klipper on Asus Tinkerboard, which is a similair board to Raspberry but with higher power. It is currently running Linaro Stretch OS (Ubuntu based). Raspbian is Debian based.

Problem is selecting the serial port. Basic ports like ttyUSB0 or ttyS0 do not wok. First one results in a general port not found and the second one keeps flashing via the port for 20 minutes without any results.
Note that this is a completely fresh octoprint installation and the printer was not connected (not even once). Could that be the problem that it is not listed in /dev/ ?

**Here is the output of dmesg of serial ports:**
`dmesg | grep tty
[    0.000000] Kernel command line: earlyprintk quiet splash plymouth.ignore-serial-consoles console=tty1 rw init=/sbin/init uboot_version=2017.07-g8c31a11215 root=/dev/mmcblk0p2
[    0.003994] console [tty1] enabled
[    0.529651] Registered FIQ tty driver
[    2.167580] ff180000.serial: ttyS0 at MMIO 0xff180000 (irq = 38, base_baud = 1500000) is a 16550A
[    2.169037] ff190000.serial: ttyS1 at MMIO 0xff190000 (irq = 39, base_baud = 1500000) is a 16550A
[    2.170187] ff1b0000.serial: ttyS3 at MMIO 0xff1b0000 (irq = 40, base_baud = 1500000) is a 16550A
[    2.171507] ff1c0000.serial: ttyS4 at MMIO 0xff1c0000 (irq = 41, base_baud = 1500000) is a 16550A
[   25.877417] ttyS0 - failed to request DMA, use interrupt mode
linaro@tinkerboard:~/klipper$ make flash FLASH_DEVICE=/dev/ttyS0
  Flashing out/klipper.elf.hex to /dev/ttyS0 via avrdude
^Csrc/avr/Makefile:32: recipe for target 'flash' failed
make: *** [flash] Interrupt
`

**Output of trying to flash on ttyUSB0:**
`linaro@tinkerboard:~/klipper$ make flash FLASH_DEVICE=/dev/ttyUSB
  Flashing out/klipper.elf.hex to /dev/ttyUSB via avrdude
avrdude: ser_open(): can't open device "/dev/ttyUSB": No such file or directory
ioctl("TIOCMGET"): Invalid argument
ioctl("TIOCMGET"): Invalid argument
vrdude: stk500v2_ReceiveMessage(): timeout
vrdude: stk500v2_ReceiveMessage(): timeout
vrdude: stk500v2_ReceiveMessage(): timeout
vrdude: stk500v2_ReceiveMessage(): timeout
vrdude: stk500v2_ReceiveMessage(): timeout
vrdude: stk500v2_ReceiveMessage(): timeout
avrdude: stk500v2_getsync(): timeout communicating with programmer
avrdude done.  Thank you.
`

**Output of serial ports on TinkerBoard:**
`
linaro@tinkerboard:~$ ls /dev/ttyS*
/dev/ttyS0  /dev/ttyS1  /dev/ttyS2  /dev/ttyS3  /dev/ttyS4
`

I might be missing something, maybe when I connect the printer atleast once it will create necessary files for me to then disconnect it and flash it. Thanks for any help. I can post anything you ask me to.

Arksine, 2018-12-17:
According to your output you are leaving the 0 off of /dev/ttyUSB0 when trying to flash.

welo78, 2018-12-17:
@Arksine sorry the outputs got a little merged, I tried both usb and usb0

Arksine, 2018-12-17:
@welo78  I see the output of /dev/ttyS*, but how about /dev/tty*?  Some boards enumerate to ttyUSB0, others ttyACM0.
I see that you are trying to flash from a Tinkerboard, but what board are you trying to flash to?

KevinOConnor, 2018-12-21:
I would follow the steps at: https://github.com/KevinOConnor/klipper/blob/master/docs/FAQ.md#wheres-my-serial-port to find the serial port.

FYI, we know Klipper runs fine on almost all Linux distros, but there's too many variations to document the installation process on each - https://github.com/KevinOConnor/klipper/blob/master/docs/FAQ.md#can-i-run-klipper-on-something-other-than-a-raspberry-pi-3

-Kevin

KevinOConnor, 2019-01-06:
I'm going to close this as I think the original question was answered.

-Kevin

closed 2019-01-06 by KevinOConnor