Klipper3d/klipper #1008: Support for Atmega 1281 (Minitronics)

BKLronin, 2018-12-18:
Hello Klippers,
I´ve got 2 printers in constant use which are running Minitronics with Atmega1281. Seems to be a minimal Version of a 1284u. I´ve red that there are some more changes neccessary with pins and all so implementing it myself sound overly time consuming. 

The board is discontinued so it might not be worth it to support but at least some people including me would be happy as those are my main printers and I already made the full config for it until I discovered that I can´t flash the firmware.

Thank you!

KevinOConnor, 2018-12-18:
As near as I can tell, the atmega1281 is the same as an atmega1280 with less pins.  You could try compiling and flashing for the atmega1280.  If avrdude complains, you could try manually running avrdude - something like: `avrdude -carduino -patmega1281 -P/dev/ttyACM0 -b57600 -D -Uout/klipper.elf.hex`

-Kevin

BKLronin, 2018-12-18:
Thanks Kevin,
I tried the 1280 but the installer geave me errors. That was some time ago so I can´t remember excatly the ouput it gave me. Trying the avrdude path now.

BKLronin, 2018-12-19:
That worked for me with 
`avrdude -carduino -patmega1281 -P/dev/ttyUSB0 -b57600 -D -Uout/klipper.elf.hex`

BKLronin, 2018-12-19:
For some reason the ar6 pin didn´t work. I switched to ar19 instead that seems to work better. In case someone else got that problem maybe because different mappings for 1281.

KevinOConnor, 2018-12-20:
I would avoid using the arduino aliases on the 1281 - those aliases will be super confusing with that chip (to the best of my knowledge, the Arduino project never released a board with the atmega1281).  You should be able to map all the pins to their hardware names following the directions at: https://github.com/KevinOConnor/klipper/blob/master/docs/FAQ.md#how-do-i-convert-a-marlin-pin-number-to-a-klipper-pin-name - but you'll need to use Marlin's fastio_1281.h file: https://github.com/MarlinFirmware/Marlin/blob/1.1.9/Marlin/fastio_1281.h

If you're able to put together a config for a board with this chip, we can add it to the Klipper config/ directory.

-Kevin

BKLronin, 2018-12-20:
I think I got it thanks to the puzzle parts you delivered. I already began to dig into the arduino IDE changes they did to find what happended but this helped. Haven´t found the pin mapping conventions in the faq. 

It is working now, I attach it here as I´m not very git approved. 
[printer.zip](https://github.com/KevinOConnor/klipper/files/2699045/printer.zip)

KevinOConnor, 2018-12-20:
Okay, thanks.  In keeping with the other configs, I'd add this as `config/generic-minitronics1.cfg` with the following content:
```
# This file contains common pin mappings for Minitronics v1.0
# boards. To use this config, the firmware should be compiled for the
# AVR atmega1280.

# The "make flash" command does not work on the Minitronics v1.0
# because the board actually has an atmega1281 chip. Use the following
# command to flash the board:
#  avrdude -carduino -patmega1281 -P/dev/ttyUSB0 -b57600 -D -Uout/klipper.elf.hex

# See the example.cfg file for a description of available parameters.

[stepper_x]
step_pin: PF2
dir_pin: PF1
enable_pin: !PF3
step_distance: .0125
endstop_pin: ^!PE3
position_endstop: 0
position_max: 200
homing_speed: 50

[stepper_y]
step_pin: PA1
dir_pin: PA2
enable_pin: !PA0
step_distance: .0125
endstop_pin: ^!PE4
position_endstop: 0
position_max: 200
homing_speed: 50

[stepper_z]
step_pin: PA4
dir_pin: !PA5
enable_pin: !PA3
step_distance: .0025
endstop_pin: ^!PB4
position_endstop: 0.5
position_max: 200

[extruder]
step_pin: PA7
dir_pin: PA6
enable_pin: !PG2
step_distance: .002
nozzle_diameter: 0.400
filament_diameter: 1.750
heater_pin: PB5
sensor_type: EPCOS 100K B57560G104F
sensor_pin: PF7
control: pid
pid_Kp: 22.2
pid_Ki: 1.08
pid_Kd: 114
min_temp: 0
max_temp: 250

[heater_bed]
heater_pin: PE5
sensor_type: EPCOS 100K B57560G104F
sensor_pin: PF6
control: watermark
min_temp: 0
max_temp: 130

[fan]
pin: PB7

[mcu]
serial: /dev/ttyUSB0

[printer]
kinematics: cartesian
max_velocity: 300
max_accel: 3000
max_z_velocity: 5
max_z_accel: 100

[static_digital_output yellow_led]
pins: PF0
```

Let me know if this looks okay to you.
-Kevin

BKLronin, 2018-12-20:
Looks good :)

KevinOConnor, 2018-12-20:
Okay - commit a5cb74d5.

-Kevin

BKLronin, 2018-12-20:
P.S: Thats my first commit, pretty happy about that. Other than that I´m glad to have one of the cartesians on klipper now and the second on marlin for direct comparison (identical machines).

KevinOConnor, 2018-12-27:
Great.  I'm going to close this issue as I think support has been resolved.

-Kevin

closed 2018-12-27 by KevinOConnor