Klipper3d/klipper #1009: Unknown command:"M108"

Hywelmartin, 2018-12-18:
So I started a print and after it got to initial temps set. it halted

 I got this from the "terminal"
```
Send: M84
Recv: ok
Send: M104 T0 S0
Recv: Not SD printing.
Send: M108Recv: ok
Send: M140 S0Recv: ok
Send: M106 S0Recv: ok
Recv: okChanging monitoring state from "Cancelling" to "Operational"
Recv: // Unknown command:"M108"
```
I have checked the gcode for M108, and there isn't any so this must be octoprint.. timer to shut of the printer or something??

[klippy.log](https://github.com/KevinOConnor/klipper/files/2690566/klippy.log)

jakep82, 2018-12-18:
It is Octoprint. I believe you can disable that in the Behavior tab.

Hywelmartin, 2018-12-18:
So now is that sorted... the 108 issue... but why did it happen in the first place??

Hywelmartin, 2018-12-18:
I wasn't even past the startup gcode... and I printed the gcode after a restart..
[ABS.rpi2-bottom_8020_netfabb.0.3.20181218-133355klipper0.6.gcode.txt](https://github.com/KevinOConnor/klipper/files/2692262/ABS.rpi2-bottom_8020_netfabb.0.3.20181218-133355klipper0.6.gcode.txt)

yesterday I got the same but that was on the famous ...... so the end code made it to go to high..(semi calculated) I add that as well
[PLA.gnomademax3flatfingersml3(1).0.2.20181217-173245klipper0.6.gcode.zip](https://github.com/KevinOConnor/klipper/files/2692279/PLA.gnomademax3flatfingersml3.1.0.2.20181217-173245klipper0.6.gcode.zip)

By the way I found an error in the "SET_DELTA_HEIGHT" gcode

> G0 F4000 X0 YO Z10
> instead of
> G0 F4000 X0 Y0 Z10

Could it be it, but I have been running with that a while...

KevinOConnor, 2018-12-18:
It looks like the g-code was in relative coordinate mode (G91) instead of the normal absolute mode (G90) at the time of the M112.  If it was also in relative mode at the time of the print start then your `G28`, `G0 ... Z10` sequence in SET_DELTA_HEIGHT would have commanded the printer up instead of down and Klipper correctly reported an error on the move (out of range move to Z=285.703).  G-code is horrible to work with - you might want to add G90 commands to your macros and start scripts.

-Kevin

Hywelmartin, 2018-12-19:
Thanks. I will do..

And have a merry x-mas

closed 2018-12-19 by Hywelmartin