Klipper3d/klipper #1010: Klipper on BIGTREETECH SKR V1.1

rmcbc, 2018-12-18:
Did anyone try klipper on BIGTREETECH SKR V1.1, the price is very good and it's an ARM processor, they say thats compatible with Smoothieware, I think they use the same pins.
I want to buy this board to use LV8729 in 1/128 or other smoother driver and need some processing power.
So I'm asking if anyone knows about this board.

KevinOConnor, 2018-12-20:
I don't have any experience with that board.  From the product description, it looks like it would work with Klipper.  However, the only way to know for sure is for someone to test it on the actual hardware.

-Kevin

rmcbc, 2018-12-20:
@KevinOConnor I bought one board from aliexpress, when it arrives (30~40 days :( ) I'm going to try to put it to work, after that create a .CFG for it.

derdigge, 2018-12-20:
Hello here i was pointed here from a Discord community. I allready run that board on a CR10 for testings. It works good so far. Just start with the smoothieboard example. I had only made prints smaller 5 hours wich came out great. Keep in mind that servo connections are slightly trick on that one if you are with a bltouch for example.

rmcbc, 2018-12-20:
@derdigge if you can please drop here your "printer.cfg" for my starting point. Thanks.

derdigge, 2018-12-21:
Its mostly the soothieboard example.

https://github.com/KevinOConnor/klipper/blob/master/config/generic-smoothieboard.cfg

rmcbc, 2018-12-21:
@derdigge but the pins on the smoothieboard.cfg don't seem to match with this board,
For example:

[stepper_x]
step_pin: P2.0  -> P0.4
dir_pin: P0.5 -> P0.5
enable_pin: !P0.4 -> P4.28

Is this what I have to do change the pin according to this image?
[image]

derdigge, 2018-12-21:
I post that config when i am at home :) . But you got it right so far. If you google for mks sgen there is an article in makerbase wiki.

KevinOConnor, 2019-01-06:
I've heard from a few different people that this board does work under Klipper.  I'm going to close this as I think the original question was answered.  If someone has the board and would like to submit a PR with an example config that would be great.

-Kevin

closed 2019-01-06 by KevinOConnor

james-golden, 2019-02-01:
Kevin, I have been able to get this board to work well with the Bigtreetech version of the Full Graphic Smart Controller.  NO SD card usage though.  I have no idea how to make a pull request at all.  Just a bit of back ground, I went from an Anycubic trigorrila to the Bigtreetech sir v1.1 board.  From trial and error, and searching thru the Issues on you GitHub, it is working great.

here is what I had to do to get the Display working 

[display]
lcd_type: st7920
cs_pin: P0.16
sclk_pin: P0.15
sid_pin: P0.18
encoder_pins: ^P3.25, ^P3.26
click_pin: ^!P2.11

Don't have the speaker hooked up because it is annoying.

Skorpi08, 2019-02-21:
Is it possible use UART instead of USB?
does it have to be crossed?
TX-RX
RX-TX

ralxnder, 2019-06-06:
Can someone point me to a source that shows how to get this board working? I've got this board and have gone through every option that I have in "make menuconfig" to generate the firmware.bin file. I copy that file over to the SD card and power up the board but I cannot connect to it through OctoPrint to save my life.