Klipper3d/klipper #1011: Repeat crashing on same model

dotps1, 2018-12-19:
I am having the same repeat crashing on the same model, despite different orientations of it.  and, its causing my entire rpi to crash so I am sure the klippy log is useless to you, i have to reboot my pi to make it responsive after the crash for me to be able to get the log, this has happened 5x times now.  i grabbed the log after the last crash, i hope you can find something in it.
[klippy.log](https://github.com/KevinOConnor/klipper/files/2693318/klippy.log)

here is the model, printing the skull when it happens: https://www.thingiverse.com/thing:1962784

really sucks, tried it a few times, gets about 80ish percent done and fails.

dotps1, 2018-12-19:
is there some way to grab the log after having to force reboot the pi?

KevinOConnor, 2018-12-20:
Alas, the log is currently stored in the /tmp/ directory which is not saved across reboots.  It's possible to modify the /etc/default/klipper to place the log in a different location.

If you're seeing raspberry pi reboots, it's likely voltage issues - check out the steps in this FAQ item: https://github.com/KevinOConnor/klipper/blob/master/docs/FAQ.md#i-keep-getting-random-lost-communication-with-mcu-errors

-Kevin

KevinOConnor, 2019-01-06:
Were you able to find anything further on this?

-Kevin

dotps1, 2019-01-08:
sorry, I haven't I've been trying to get my BLTouch working for the past week.  Ill try to get to it by this weekend.

Thanks.

KevinOConnor, 2019-01-16:
I'm going to close this issue - I'll reopen if it recurs.

-Kevin

closed 2019-01-16 by KevinOConnor