Klipper3d/klipper #1013: Resetting multiple MCUs via command fails

yschroeder, 2018-12-19:
I have 4 MCUs connected to Klipper. When I change the config, I reset everything with the `RESTART` command. This reliably fails to reset all MCUs. On the first `RESTART` it fails to reset the first MCU, on the second try it resets the first and fails on the second MCU, on the third try it resets MCUs 1 and 2 but fails on the third, on the fourth try it fails on the fourth MCU. Only on the fifth try it resets all 4 MCUs and Klipper reports `ready`.

I expect it to reset all MCUs on the first try.

One of the errors copied from the log:
```
MCU error during connect
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/klippy.py", line 140, in _connect
    cb('connect')
  File "/home/pi/klipper/klippy/mcu.py", line 768, in printer_state
    self._connect()
  File "/home/pi/klipper/klippy/mcu.py", line 623, in _connect
    self._check_config()
  File "/home/pi/klipper/klippy/mcu.py", line 581, in _check_config
    self._name,))
error: Failed automated reset of MCU 'axis_d'
Failed automated reset of MCU 'axis_d'
```

[klippy_resetproblem.log](https://github.com/KevinOConnor/klipper/files/2693820/klippy_resetproblem.log)

KevinOConnor, 2018-12-21:
Okay, thanks.  On a config change the klipper host software has to reset each micro-controller.  It detects the need for a reset after connecting to the first micro-controller, but when it then goes on to perform a restart, the other micro-controllers have not yet performed their identification process - so the "reset" command for these micro-controllers isn't yet known.  So, no way to reset the other mcus.   (By the fifth attempt, all the micro-controllers have the updated config, so there is no longer a need to perform the restart.)  I'll have to think on how to address this.

-Kevin

KevinOConnor, 2019-11-05:
PR #2144 should fix this.

-Kevin

PR #2144 (2019-11-06): Fix for multi-mcu FIRMWARE_RESTART bug
This should hopefully fix #1013.

@vladbabii - fyi

-Kevin
closed 2019-11-06 by KevinOConnor