Klipper3d/klipper #1014: 4x Klipper on one pi 3B+

Fleckenzwerg94, 2018-12-19:
i have 4x Ocotorpint on one pi: https://www.youtube.com/watch?v=7Saa1HpLRoM&t

Can i run 4x klipper? And if yes, how?

AxMod3DPrint, 2018-12-19:
Technically, yes, not recommended though as the Pi will not have enough
processing power.

jakep82, 2018-12-19:
See Kevin's response on this issue.  It doesn't sound like it's a simple thing to do.

https://github.com/KevinOConnor/klipper/issues/961

yschroeder, 2018-12-20:
My guess would be, that you only need to tell the different Klipper instances to load different config files, write to different logfiles and create different virtual serial ports (`/tmp/printer`). These three things can be done by changing the command line arguments of klippy, see `klippy/klippy.py`. To create a second Klipper service, copy the systemd service file and modify it accordingly.

This is what I would try, but no guarantee that this works.

AxMod3DPrint, 2018-12-20:
@yschroeder has it correct. However, I'll re-iterate that I wouldn't do it on a Pi. I jave 2 Klipper instances running on a ZBox which has more grunt. Also, If you're unfamiliar with Linux it can be difficult.

KevinOConnor, 2018-12-27:
I'm going to close this as I think the original question was answered (or at least, as best as it can be answered).

-Kevin

closed 2018-12-27 by KevinOConnor