Klipper3d/klipper #1017: Beaglebone has timing issue with master branch.

popshansen, 2018-12-19:
The master branch does not work on beaglebone black when homming, it home the X and the stops with internal G28 error, MCU 'mcu' shutdown: Timer too close.
I have a git version "v0.6.0-663-g6df7356-dirty" that works perfectly. No problems with homming.
Just to inform you that the master branch has timing issue on the beaglebone black.

Best regards

[klippy.log](https://github.com/KevinOConnor/klipper/files/2695467/klippy.log)

BlackStump, 2018-12-19:
I have to change to
HOMING_STEP_DELAY=0.000001
in /klipper/klippy/homing.py
to get homing to work

popshansen, 2018-12-20:
Thanks.
It works when I change that value. I can see in the "old" code that it also have 0.000001 and 0.00000025 is comment out.

closed 2018-12-20 by popshansen